Klipper3d/klipper #746: Internal error on command:"G28"

cassianoleal, 2018-10-05:
Since commit b0d859f220457f0a1c15b5be452620f19043ae16 I haven't been able to home my printer as a G28 crashes the host software.

### Output of terminal:

```
Send: G28
Recv: // Klipper state: Shutdown
Recv: !! Internal error on command:"G28"
Recv: !! Internal error on command:"G28"
Recv: ok
```

#### klippy.log

```
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 188, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 69, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 21, in cmd_G28
    self.gcode.cmd_G28(params)
  File "/home/pi/klipper/klippy/gcode.py", line 520, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/homing.py", line 116, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 101, in home
    self._home_axis(homing_state, axis, self.rails[axis])
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 85, in _home_axis
    second_homing_speed, second_home=True)
  File "/home/pi/klipper/klippy/homing.py", line 112, in home
    verify_movement=second_home and self.verify_retract)
  File "/home/pi/klipper/klippy/homing.py", line 67, in _homing_move
    mcu_endstop.home_wait(move_end_print_time)
  File "/home/pi/klipper/klippy/mcu.py", line 195, in home_wait
    while self._check_busy(eventtime, home_end_time):
  File "/home/pi/klipper/klippy/mcu.py", line 223, in _check_busy
    raise error("MCU is shutdown")
error: MCU is shutdown
Dumping gcode input 4 blocks
Read 2872.609575: 'M105\n'
Read 2876.573820: 'M105\n'
Read 2881.581556: 'M105\n'
Read 2882.723751: 'G28\n'
gcode state: absolutecoord=True absoluteextrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[198.0, 112.0, 0.0, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=100.0
Internal error on command:"G28"
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 188, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 69, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 57, in cmd_G28
    self.gcode.run_script_from_command(self.script)
  File "/home/pi/klipper/klippy/gcode.py", line 264, in run_script_from_command
    self.process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 188, in process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 69, in <lambda>
    func = lambda params: origfunc(self.get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 21, in cmd_G28
    self.gcode.cmd_G28(params)
  File "/home/pi/klipper/klippy/gcode.py", line 520, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/homing.py", line 116, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 101, in home
    self._home_axis(homing_state, axis, self.rails[axis])
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 85, in _home_axis
[... 192 lines cut ...]
Receive: 77 2889.622585 2888.884695 14: seq: 1a, analog_in_state oid=15 next_clock=274400000 value=7835
Receive: 78 2889.831498 2888.884695 14: seq: 1a, analog_in_state oid=6 next_clock=277760000 value=7802
Receive: 79 2889.870424 2889.866922 11: seq: 1b, clock clock=273696720
Receive: 80 2889.893124 2889.890462 9: seq: 1c, end_stop_state oid=5 homing=1 pin=1
Receive: 81 2889.921866 2889.890462 14: seq: 1c, analog_in_state oid=15 next_clock=279200000 value=7836
Receive: 82 2890.131971 2889.890462 14: seq: 1c, analog_in_state oid=6 next_clock=282560000 value=7792
Receive: 83 2890.222075 2889.890462 14: seq: 1c, analog_in_state oid=15 next_clock=284000000 value=7836
Receive: 84 2890.432220 2889.890462 14: seq: 1c, analog_in_state oid=6 next_clock=287360000 value=7802
Receive: 85 2890.440007 2889.890462 9: seq: 1c, end_stop_state oid=5 homing=0 pin=0
Receive: 86 2890.497637 2890.494840 9: seq: 1d, stepper_position oid=3 pos=-1998
Receive: 87 2890.521357 2890.494840 14: seq: 1d, analog_in_state oid=15 next_clock=288800000 value=7835
Receive: 88 2890.732230 2890.557482 14: seq: 12, analog_in_state oid=6 next_clock=292160000 value=7801
Receive: 89 2890.822346 2890.557482 14: seq: 12, analog_in_state oid=15 next_clock=293600000 value=7839
Receive: 90 2890.856301 2890.852616 11: seq: 13, clock clock=289469266
Receive: 91 2891.031349 2890.852616 14: seq: 13, analog_in_state oid=6 next_clock=296960000 value=7803
Receive: 92 2891.121488 2890.852616 14: seq: 13, analog_in_state oid=15 next_clock=298400000 value=7838
Receive: 93 2891.331658 2890.852616 14: seq: 13, analog_in_state oid=6 next_clock=301760000 value=7803
Receive: 94 2891.421710 2890.852616 14: seq: 13, analog_in_state oid=15 next_clock=303200000 value=7837
Receive: 95 2891.560885 2891.557390 9: seq: 14, end_stop_state oid=5 homing=1 pin=1
Receive: 96 2891.631950 2891.557390 14: seq: 14, analog_in_state oid=6 next_clock=306560000 value=7802
Receive: 97 2891.722115 2891.557390 14: seq: 14, analog_in_state oid=15 next_clock=308000000 value=7838
Receive: 98 2891.838567 2891.835569 11: seq: 15, clock clock=305196124
Receive: 99 2891.854898 2891.835569 12: seq: 15, shutdown clock=305438646 static_string_id=38
Internal error on command:"G28"
```

cassianoleal, 2018-10-05:
If it helps, I have this set in my `printer.cfg`:

```
[homing_override]
gcode: G28 X0Y0
  G1 X198 Y112 F6000
  G28 Z0
axes: z
set_position_x: 150
set_position_y: 110
set_position_z: 0
```

KevinOConnor, 2018-10-07:
You'll need to attach the full klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

cassianoleal, 2018-10-07:
Hmm... Ok. I did look at the log and thought I was posting everything potentially useful but I'll reproduce it again and post the rest.

Out of curiosity, and so I have more tools in the future when troubleshooting, what other kind of info are you looking for in the logs? IIRC it was just the printer.cfg dump and then the `Status` lines.

cassianoleal, 2018-10-08:
There you go, full log from `systemctl start klipper` to `systemctl stop klipper`.

```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-509-g909eef3'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Mon Oct  8 00:32:24 2018 (1538958744.1 238.6)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = !ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 300
homing_speed = 150

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 220
homing_speed = 120

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .0025
endstop_pin = probe:z_virtual_endstop
position_max = 270
homing_speed = 800

[probe]
pin = ^!ar18
x_offset = -48.0
y_offset = -2.0
z_offset = 2.475
speed = 5.0

[extruder]
step_pin = ar26
dir_pin = !ar28
enable_pin = !ar24
step_distance = 0.0106
nozzle_diameter = 0.4
filament_diameter = 1.75
max_extrude_cross_section = 1.5
max_extrude_only_distance = 200
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 29.865
pid_ki = 1.933
pid_kd = 115.353
min_temp = 0
max_temp = 285
pressure_advance = 0.825

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
[... 361 lines cut ...]
  File "/home/pi/klipper/klippy/homing.py", line 116, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 101, in home
    self._home_axis(homing_state, axis, self.rails[axis])
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 85, in _home_axis
    second_homing_speed, second_home=True)
  File "/home/pi/klipper/klippy/homing.py", line 112, in home
    verify_movement=second_home and self.verify_retract)
  File "/home/pi/klipper/klippy/homing.py", line 67, in _homing_move
    mcu_endstop.home_wait(move_end_print_time)
  File "/home/pi/klipper/klippy/mcu.py", line 195, in home_wait
    while self._check_busy(eventtime, home_end_time):
  File "/home/pi/klipper/klippy/mcu.py", line 223, in _check_busy
    raise error("MCU is shutdown")
error: MCU is shutdown
Internal error on command:"G28"
Stats 269.3: gcodein=77 mcu: mcu_awake=0.015 mcu_task_avg=0.000136 mcu_task_stddev=0.000147 bytes_write=5849 bytes_read=7197 bytes_retransmit=0 bytes_invalid=0 send_seq=224 receive_seq=224 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000670 heater_bed: target=0 temp=24.0 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Stats 270.3: gcodein=77 mcu: mcu_awake=0.012 mcu_task_avg=0.000114 mcu_task_stddev=0.000116 bytes_write=5855 bytes_read=7311 bytes_retransmit=0 bytes_invalid=0 send_seq=225 receive_seq=225 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000672 heater_bed: target=0 temp=24.0 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
Stats 271.3: gcodein=77 mcu: mcu_awake=0.012 mcu_task_avg=0.000114 mcu_task_stddev=0.000116 bytes_write=5861 bytes_read=7417 bytes_retransmit=0 bytes_invalid=0 send_seq=226 receive_seq=226 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000671 heater_bed: target=0 temp=24.0 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
Stats 272.3: gcodein=77 mcu: mcu_awake=0.012 mcu_task_avg=0.000114 mcu_task_stddev=0.000116 bytes_write=5867 bytes_read=7523 bytes_retransmit=0 bytes_invalid=0 send_seq=227 receive_seq=227 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000671 heater_bed: target=0 temp=23.8 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
Stats 273.3: gcodein=82 mcu: mcu_awake=0.012 mcu_task_avg=0.000114 mcu_task_stddev=0.000116 bytes_write=5873 bytes_read=7616 bytes_retransmit=0 bytes_invalid=0 send_seq=228 receive_seq=228 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000670 heater_bed: target=0 temp=23.9 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
Stats 274.4: gcodein=82 mcu: mcu_awake=0.012 mcu_task_avg=0.000114 mcu_task_stddev=0.000116 bytes_write=5879 bytes_read=7722 bytes_retransmit=0 bytes_invalid=0 send_seq=229 receive_seq=229 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000675 heater_bed: target=0 temp=23.9 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.1 pwm=0.000
Stats 275.4: gcodein=82 mcu: mcu_awake=0.006 mcu_task_avg=0.000069 mcu_task_stddev=0.000056 bytes_write=5885 bytes_read=7839 bytes_retransmit=0 bytes_invalid=0 send_seq=230 receive_seq=230 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000676 heater_bed: target=0 temp=23.9 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.4 pwm=0.000
Stats 276.4: gcodein=82 mcu: mcu_awake=0.006 mcu_task_avg=0.000069 mcu_task_stddev=0.000056 bytes_write=5891 bytes_read=7932 bytes_retransmit=0 bytes_invalid=0 send_seq=231 receive_seq=231 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000675 heater_bed: target=0 temp=23.9 pwm=0.000  print_time=262.599 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
```

KevinOConnor, 2018-10-08:
This was due to an error resulting from the unrealistic z homing_speed in your config.  It should be fixed now (commit 36ae433b).  (Though, I'd still recommend removing the unrealistic z homing_speed.)

Separately, in the future, please **attach** the klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

cassianoleal, 2018-10-08:
Agh... Apologies, I wasn't aware that attaching was a feature of Github. Should have RTFM I guess. :)

And yes, you're probably correct about my Z homing speed. I'll play around with it when I'm back at home.

Thanks!

cassianoleal, 2018-10-09:
Yup, confirmed working. Thanks! :)

closed 2018-10-09 by cassianoleal