Klipper3d/klipper #747: Bl touch repetability test

Thesecky, 2018-10-06:
Marlin has a repetability test for the bl touch. Does klipper have anything similar and if not, what comands can i use to calculate it myself.

derdigge, 2018-10-07:
@Thesecky 

I use a macro for it. Not the same but you can get an idea how ecxact your probe is. Just move your probe to desired position and issue a M48.

`[gcode_macro M48]
gcode:
	G1 Z5
	PROBE
	G1 Z5
	PROBE
	G1 Z5
	PROBE
	G1 Z5
	PROBE
	G1 Z5
	PROBE
	G1 Z20`

Thesecky, 2018-10-13:
This is what i get.
[image]
Are those results supposed to be that good, i would expect at least some difference between the measurments.

derdigge, 2018-10-15:
It looks to good to be true. Maybe our bltouch is a special ninja one? 👍 
What about increasing probe speed and see whats then?

KevinOConnor, 2018-10-15:
FYI, the inductive probe on my delta printer gets the same results when I probe the same spot.  The probe should be returning identical results.

-Kevin

derdigge, 2018-10-16:
@KevinOConnor Sorry for spaming. Which probe do you use at what probe speed?

KevinOConnor, 2018-10-16:
It's a pretty standard 12V inductive probe.  I don't have the part number handy, but it looks like https://reprap.org/mediawiki/images/thumb/2/23/Inductive-sensor-blue.jpg/300px-Inductive-sensor-blue.jpg

To be clear, this probe isn't useful on my delta printer - I get different readings at different parts of the bed because minor differences in arm length result in a tilt to the effector.  That said, if I probe the same location multiple times in a row I generally get the same exact reading from the inductive probe.

I use the default probing speed of 5mm/s.

-Kevin

derdigge, 2018-10-16:
Thanks. I use the same in a 24v system. So the probe is powered with 24v. But i have small fluctuation in the probings ~ 0.01 - 0.04. I probe with 2mm/s. Do you power it with 12v? Maybe i should try it

KevinOConnor, 2018-10-16:
Yes - 12V.  It's not unusual for me to see one step difference, but it's pretty stable for me.  I just ran a quick test (my step distance is .0125 so the difference below is just one step):
```
Recv: // probe at -2.028,-2.138 is z=0.370724
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.370724
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.358224
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.370724
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.370724
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.370724
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.358224
Send: G1 Z2
Recv: // probe at -2.028,-2.138 is z=0.370724
```

closed 2018-10-23 by KevinOConnor