Klipper3d/klipper #750: Duet 2 Meastro

Grimpster, 2018-10-06:
Ok not really sure if it's spose to be an issue or a request? Anyways, here goes.

From what I understand the Duet 2 line is supported by klipper as of now.

I haven't really been able to dig up the pin layout in the klipper files, nor could I find it for Duet wifi or Maestro.

Lets just pretend that I install klipper on the Maestro, how far off an wifi do you think it would be?
Or do we know of anyone working on this port ? Cause I kind of feel klipper is my one and only right now. And I can't be arsed to get into RRF, and I cant find anything to compare it with.

KevinOConnor, 2018-10-07:
The Duet 2 Maestro does look like an interesting board.  However, as I understand it, it would require software development work to support.  (It's a different sam4 cpu, the tmc2224s are behind an analog mux, the current sam4 code needs usb support in general.)

-Kevin

Grimpster, 2018-10-07:
Goddammit, I guess it's bye bye klipper then..

 But the the duet wifi is working, I don't k ow of the sam4 is that much different, I know nothing on coding this.

And the pin layout, so I guess some guru can make it work if, but I guess it's easier said than done.

I don't feel like going RRF
I guess I was under the impression that it would work or was easier to set up by porting from Duet. Ahh well.

FHeilmann, 2018-10-08:
Maestro is SAM4S while the Duet is SAM4E, while there are a lot of similarities, someone who has access to a board would have to work on the port, as it doesn't make sense for someone who doesn't have access to a physical board to do the dev work.

Further, as Kevin mentioned, the USB support is still missing for the SAM4E as well.

Grimpster, 2018-10-08:
Yeah I get that it's hard to work in the blind heh. 

I'm not really sure what the USB support does, I guess it's to send and receive commands.

Ill mount the maestro and try out the RRF if it goes south, I'll go back and try and dual ramps I guess. 

Cheers.

FHeilmann, 2018-10-08:
As of now, to use the SAM4E (like the one on the duet wifi) you have to connect a serial to usb converter to one of the boards serial ports (thereby losing either paneldue or wifi support). 

The USB support allows to simply connect the USB cable to the board and achieve communication that way.

Grimpster, 2018-10-08:
Ah, yeah I guess the usb is just 5v in atm.

I dont use a display on my printer, I get all I need from octoprint and my wifi needs are nonexistant already. So I can live with that. 

Sadly I cannot contribute with the porting/programming only hardware related stuff.
Well see what happens, I'm sure I'm not the only one getting this board ;)

KevinOConnor, 2018-10-16:
It looks like the original question was answered, so I'm going to close this issue.

closed 2018-10-16 by KevinOConnor