Klipper3d/klipper #751: (enhancement) bed_mesh.py

aeropic, 2018-10-06:
@Arksine : I have a servo probe installed on my ANET A8. As described in #616 I need to suppress the forced homing inside bed mesh leveling. Up to now I did it just commenting the adhoc line in the code.
However this is not convenient in case of new releases...
I would propose a simple mod coded in the linked files:
- in printer.cfg: added a section :
with_homing: NO
# includes or not a G28 homing command.  Default is "YES". 
# In case of a servo probe it may be usefull not to perform the homing 
# inside the bed_mesh in that case set this field to "NO"

in bed_mesh.py: added the code to read the 'with_homing' config and added the according code to skip or not the G28 command.
By default the code works as today main branch.

Do you think this could merged into the main branch ? As I'm quite noob under github and python, I do not know how to do this kind of request/proposal ...

And thanks again for the nice SAVE_config feature coupled with bed_mesh!

[bed_mesh.py.txt](https://github.com/KevinOConnor/klipper/files/2453299/bed_mesh.py.txt)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2453300/printer.cfg.txt)

KevinOConnor, 2018-10-07:
I don't think we need two github issues to track this.  Lets use one or the other.

FWIW, I'd be inclined to remove the G28 from bed_tilt_calibrate, delta_calibrate, and bed_mesh_calibrate.  Users can issue the G28 themselves or add a gcode_macro that does it for them.

-Kevin

aeropic, 2018-10-08:
You're fully right Kevin, I closed the #616 which was addressing mainly the servo probe aspect. 
Now this one remains : we need a way to get rid of the G28 from bed_tilt_caibrate, delta_caibrate and bed_mesh_calibrate.
Indeed using gcode macros is IMHO the best way to proceed.
Today, with servo probes or any other mechanically activated probe there is no other way to proceed that activating/deactivating the probe at each point which is a mess in terms of speed and mechanical wear ...

I would too prefer a simple removal of G28 from bed_mesh but the way I did it in this post allows to keep the two worlds ;-)

aeropic, 2018-10-22:
So cool, many thanks for this feedback!
I guess we can close this issue now ?

closed 2018-10-24 by aeropic