Klipper3d/klipper #754: [Feature Request] Generic endstop interface

FHeilmann, 2018-10-07:
Hi @KevinOConnor,
I am currently working on bringing up an i2c to gpio expander to work with klipper. Naturally, I want to be able to use the gpio pins of this expander like the other gpio pins of the mcu.

I was looking at the implementation of the endstops in `endstop.c` and realized that I would have to duplicate a lot of code, just to change 1 or 2 lines per function. This seems wasteful to me, and while it is the easier approach, I believe the cleaner approach would be to generalize the gpio interface, such that things like endstop and (scheduled) gpio handling could be done using a unified set of commands.

To give an idea of what I mean, I pushed a set of changes here:
https://github.com/FHeilmann/klipper/commit/ebae05140a2de9ded3836eeda322f56ebf30e04d
to make the endstop interface as generic as possible.
This is done by removing the specifics of the `gpio_in` struct and the `gpio_in_read` function from most of the functions and declaring a generic `end_stop` struct.

This implementation doesn't work as of now (probably because I'm really bad at juggling pointers, the endstops report as always triggered) but it should give an idea about what I want to do.

In order for me to implement the i2c gpio expander, I would only have to implement a new `command_config_i2c_end_stop` and a new `i2c_endstop_read` function instead of having to duplicate all the code in `endstop.c`

Of course, if you want to keep your code clean of such things I will do so. Otherwise I'd appreciate if you looked into this or gave me some pointers on how to achieve the goal.

Bests,

Florian

KevinOConnor, 2018-10-07:
Hi Florian,

Does one of your boards use a gpio expander for endstops?  That's unfortunate and will be difficult to support well.  The problem isn't the endstop code, but handling of irq/task context switching.  The software timer events are called with irqs disabled and i2c accesses aren't particularly fast.  They're definitely much greater than the typical 1-5us that we expect a software timer to finish in.  Worse, if anything uses the i2c port from irq context then all of the i2c code would likely need to be run with irqs disabled.  That could add notably to irq handling latency.

I've no issue making the endstop code more generic (assuming we can solve the above).  As a minor change, I'd recommend against using function pointers - they add overhead even on mcus that don't support the i2c expanders.  It should be straight-forward to introduce a generic read_endstop() method that walks a series of if's to determine which low-level function to call.

-Kevin

FHeilmann, 2018-10-08:
Hi Kevin,
thanks for your detailed response. The points you make are valid, and after some more thinking I think I've found a workaround for my specific board (duet wifi with duex expansion board (endstops 5-9 on the duex are handled via an sx1509 i2c to gpio expander)), much like you did with the TMC2130 virtual endstops. To that end, I have a small unrelated question: How does klipper home a delta? The reason for my asking is the following:

I can work around the irq issue by using the `interrupt-out` of the sx1509. I would do so by marking only the pin of interest for interrupts before homing, and then treat the interrupt-out as if it were the endstop pin of the main MCU. However, if the mcu tries to home multiple endstops which are connected to the expander at the same time (which I expect it would on a delta if my understanding is correct) then I might have to do some additional handling.

I'll close this issue, but I'd appreciate your continued feedback.

Bests,

Florian

closed 2018-10-08 by FHeilmann

KevinOConnor, 2018-10-08:
> How does klipper home a delta?

All three motors are homed simultaneously and each motor individually stops when the endstop for that tower is triggered.  The same would occur if one homes multiple Z steppers that each have their own endstop.

> I can work around the irq issue by using the interrupt-out of the sx1509.

That sounds like it would work as long as there is only one stepper using that sx1509 at any given time.
But, you might want to check how well the sx1509 does noise detection before raising the interrupt line (the endstop.c code samples multiple times before signalling to avoid spurious signals).

Homing multiple axes simultaneously via the sx1509 would be quite the pickle.  It's certainly possible but it sounds like a large amount of work.

-Kevin