Klipper3d/klipper #756: [FR] serial communication interface

derhopp, 2018-10-07:
Hi @KevinOConnor,

klipper already supports communication via SPI.  Is it possible to also include serial uart communication the same way with commands for
- Configuration with port, baud rate, rx-buffersize
- Sending a string
The rx buffer should be automatically send to console on buffer full or received "\n".

It could be the basis to include additional peripheral.  E.g. it could be used to include the basic functions of the Anycubic Mega display (like axis movement, temperature setting, cooling, kill, temptation read back, filament load) through an octoprint module or through klippy .

Best regards
Christian Hopp

KevinOConnor, 2018-10-08:
It's certainly possible to implement this.  It's not something I plan to work on, but I'm open to merging support if someone does the work.

-Kevin

derhopp, 2018-10-08:
Okay... I will check if there is a quite portable solution possible, but serial need much platform dependant code.  Otherwise I make a specific implementation for the Anycubic Mega I3 in a separate fork, like I did for Marlin.

I'll keep you informed.

KevinOConnor, 2018-10-16:
Okay - I'm going to close this for now.  Will reopen if you have further questions.

closed 2018-10-16 by KevinOConnor