Klipper3d/klipper #757: "make" failed to complete

BlaqOps, 2018-10-08:
After my Raspberry Pi Model 3 B+ failed to start due to a corrupt SD card, I did a full OS reinstall. Now I'm experiencing trouble building the firmware to flash the mcu. This is the on the latest version of OctoPi. Output is as follows:
`pi@octopi:~/klipper $ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.6.0-509-g909eef3-20181008_110803-octopi
  Linking out/klipper.elf
out/src/basecmd.o: In function `command_identify':
/home/pi/klipper/src/basecmd.c:323: undefined reference to `__mulsi3'
/home/pi/klipper/src/basecmd.c:323: undefined reference to `__mulsi3'
out/src/gpiocmds.o: In function `soft_pwm_shutdown':
/home/pi/klipper/src/gpiocmds.c:241: undefined reference to `__mulsi3'
out/src/thermocouple.o: In function `thermocouple_task':
/home/pi/klipper/src/thermocouple.c:101: undefined reference to `__bswapsi2'
/home/pi/klipper/src/thermocouple.c:119: undefined reference to `__bswapsi2'
/home/pi/klipper/src/thermocouple.c:138: undefined reference to `__bswapsi2'
/usr/lib/gcc/avr/5.4.0/../../../avr/lib/avr6/libc.a(exit.o): In function `exit':
(.text.avr-libc+0x2): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
Makefile:87: recipe for target 'out/klipper.elf' failed
make: *** [out/klipper.elf] Error 1
`

povlhp, 2018-10-08:
If you google, you will see this error is due to a missing stdlib.
Example search result:
https://github.com/riscv/riscv-gnu-toolchain/issues/267

Are you compiling for PRU ? It has the -nostdlib option in Makefile.

BlaqOps, 2018-10-08:
@povlhp No, I'm trying to compile for the Trigorilla board in the Anycubic Kossel Linear Plus. It's a Atmega2560. I had a working system before the last installation became corrupted, but a much harder going this time around.

KevinOConnor, 2018-10-08:
That's an odd error.  Try running `make clean ; make menuconfig ; make` to see if the error clears.  If that doesn't help, try rerunning the `./klipper/scripts/install-octopi.sh` script again and then the above steps.  If that doesn't help it is likely a corrupt octopi installation - try downloading the image again and reflash it to an sd card.

-Kevin

closed 2018-10-16 by KevinOConnor

aee74, 2021-08-21:
`pi@octopi:~/klipper $ make
  Creating symbolic link out/board
  Compiling out/src/sched.o
make: arm-none-eabi-gcc: Command not found
make: *** [Makefile:64: out/src/sched.o] Error 127`

any idea on what I'm missing

kmcken, 2021-08-24:
@aee74 I just got the same Error 127

ttm2112, 2021-08-25:
I have the same exact problem on Pi 4B

pi@octopi:~/klipper $ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
make: arm-none-eabi-gcc: Command not found
make: *** [Makefile:64: out/src/sched.o] Error 127
pi@octopi:~/klipper $

**menuconfig options**

(Top)
                         Klipper Firmware Configuration
[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F446)  --->
    Bootloader offset (32KiB bootloader)  --->
    Clock Reference (12 MHz crystal)  --->
    Communication interface (USB (on PA11/PA12))  --->
    USB ids  --->                    Quit
[ ] Specify a custom step p
()  GPIO pins to set at mic   Save configuration?   W)

                             (Y)es  (N)o  (C)ancel

**And i think it has something to do with it not running the .sh part correctly.**

pi@octopi:~/klipper $ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
pi@octopi:~/klipper $

I have re-imaged several times, tried Putty and Tera Term, same result. Any ideas?
Thanks in advance.

Drew3164, 2021-09-05:
Any fixes on this I have the same problem?

GITRDUN75, 2021-09-09:
Same problem here. Apparently something has been changed and the old installation instructions are now out of date and missing a new step. I miss the days when your software didnt get updated automatically every other day and screw up half your system. This modern way of doing things is a mess. I thought we left DOS in the '90s, and yet here we are having to use command line operating systems again. Seems like a step backwards for the average user to me. I do appreciate the time and effort invested by the developers at their own expense.

kmcken, 2021-09-10:
@ttm2112 is right. For me, the problem was that I was not running 
`$ ./klipper/scripts/install-octopi.sh`
correctly. 

Make sure that the .sh install works correctly, and then the rest should install properly.

CASHatFOLSOM, 2021-09-11:
`sudo apt update --allow-releaseinfo-change`
did it for me as well, thanks.

ChrisAFRcrew, 2021-09-14:
I am having the same Issue and I have ran the update and still get the error

JamesMijoc, 2021-10-04:
This worked for me, thank you.

maxpcuser, 2021-10-05:
jesus glad i found this it was driving me nuts

brokenarrow03, 2021-11-05:
I just used this thread to fix my exact issue! Thank you!

GDRPrint, 2021-11-17:
> I have the same exact problem on Pi 4B
> 
> pi@octopi:~/klipper $ make Creating symbolic link out/board Building out/autoconf.h Compiling out/src/sched.o make: arm-none-eabi-gcc: Command not found make: *** [Makefile:64: out/src/sched.o] Error 127 pi@octopi:~/klipper $
> 
> **menuconfig options**
> 
> (Top) Klipper Firmware Configuration [*] Enable extra low-level configuration options Micro-controller Architecture (STMicroelectronics STM32) ---> Processor model (STM32F446) ---> Bootloader offset (32KiB bootloader) ---> Clock Reference (12 MHz crystal) ---> Communication interface (USB (on PA11/PA12)) ---> USB ids ---> Quit [ ] Specify a custom step p () GPIO pins to set at mic Save configuration? W)
> 
> ```
>                          (Y)es  (N)o  (C)ancel
> ```
> 
> **And i think it has something to do with it not running the .sh part correctly.**
> 
> pi@octopi:~/klipper $ ./klipper/scripts/install-octopi.sh
> 
> ###### Running apt-get update...
> Hit:1 http://archive.raspberrypi.org/debian buster InRelease Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB] Reading package lists... Done E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable' N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details. pi@octopi:~/klipper $
> 
> **I ran it again and got this**
> 
> pi@octopi:~ $ ./klipper/scripts/install-octopi.sh
> 
> ###### Running apt-get update...
> [sudo] password for pi: Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB] Get:2 http://archive.raspberrypi.org/debian buster/main armhf Packages [378 kB] Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB] Reading package lists... Done E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable' N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details. pi@octopi:~ $
> 
> I have re-imaged several times, tried Putty and Tera Term, same result. Any ideas? Thanks in advance.
> 
> **One more thing. When i try to remove klipper and try again, I get this...**
> 
> pi@octopi:~ $ sudo apt-get purge remove Klipper Reading package lists... Done Building dependency tree Reading state information... Done E: Unable to locate package remove E: Unable to locate package Klipper pi@octopi:~ $
> 
> ===================== **This fixed my issues. Run before getting klipper.**
> 
> sudo apt update --allow-releaseinfo-change
> 
> .sh piece worked after that. Hope this helps someone out there.

Sure helped me, thank you for posting !!

GDRPrint, 2021-11-17:
Very Helpful , check this solution if you have that error msg

jordylane, 2021-11-22:
Sometimes you feel like a dingus, and then later you realise there are a collection of other dinguses out there being a dingus right there with you. And that makes you feel a little less like a dingus. Thanks for easing my frustration.

Targyen, 2022-01-06:
Ok, it looks like im to dumb to do this. Im not sure that i have name of my port correct because I keep gething Error 1.

pi@octo:~/klipper $ ls /dev/serial/by-id/* 

 /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

How should FLASH command be writen?

Sineos, 2022-01-06:
> How should FLASH command be writen?

Please do not simply revive some old and closed threads. Post your issue together with relevant information (klippy.log, board, printer type etc) on https://klipper.discourse.group