Klipper3d/klipper #758: Motors on, fans on. Possible?

swatti87, 2018-10-08:
Is it possible to turn on fans when motors are active?

I have a bit of an odd fan-setup but it keeps the noise down.

Block cooler fires off when heater active and shuts down when temp is below 90. This one works.
[Fan] is clearly the part cooler, controlled via Cura, based on layers.

Now i need 2 more. Or atleast one.

When motors on, turn on "stepper-fan" - with max power below 1.0 ( the noise)

When power on, turn on motherboard/driver fan - with max power below 1.0

Those last two are over-sized fans with the idea of running them at half-power for good cooling and low noise. I prefer to control these separate from each other but their power can be static. Only the part-cooler needs a variable speed.

It seems when i make a new "[fan]" or change the name of the default part-cooler, my klippy goes crazy.

mcmatrix, 2018-10-08:
Yes, it's possible.
I'm using this setup for controller fans.
Please see https://github.com/KevinOConnor/klipper/issues/529#issuecomment-411653844

swatti87, 2018-10-08:
> Yes, it's possible.
> I'm using this setup for controller fans.
> Please see [#529 (comment)](https://github.com/KevinOConnor/klipper/issues/529#issuecomment-411653844)

Well, better than nothing but i'd prefer to control them individually. At ~50% power my driver fan already screams like crazy but at the same power my stepper coolers dont do much.

Interesting idea. 
Umm... I've had some issues finding the right pin for several things. Could i simply add "all" possible pins like that, then start taking them off one-by-one when i get it going?

swatti87, 2018-10-08:
Can i do "multiple" code-snippets and add power-variables to them?

[multi_pin my_enable_pin]
pins: !ar25 

ar25 being z-stepper enable
ar13 being the "yuuuge" driver fan, should not run higher than 0.3 power
ar10 is the stepper blowers. At 0.8 power they become silent
ar11 is the part-coolers. They can run at 100% as i can control them via Cura

KevinOConnor, 2018-10-08:
I suspect most people use a [heater_fan] section even for stepper driver and board coolers (under the assumption that the motors typically only step when the extruder is on).  An alternative is to add a thermistor and use a [temperate_fan] section.  Both sections are described in config/example-extras.cfg .

-Kevin

jakep82, 2018-10-08:
I do what Kevin describes.  My driver cooling fan turns on when my extruder is active.  An alternative approach is to define a run-time configurable fan and add lines to your start and end g-code to turn it on and off.  Either approach will work with PWM.

jakep82, 2018-10-09:
Please read example-extras.cfg.  You can set up as many fans as you want, each with it's own PWM value.  Also, your description is in poor taste and unnecessary.

Hywelmartin, 2018-10-09:
[output_pin fan22]
pin = arXX
pwm = True
cycle_time = 0.0100
hardware_pwm = True
value = 0.05
scale = 255
shutdown_value = 1
#Use The fan22 port with "SET_PIN PIN=fan22 VALUE=xxx(0-255, off-max)"

cosminr86, 2018-10-12:
I just config my setup. All working good.
I put "SET_PIN PIN=fan22 VALUE=xxx(0-255, off-max)" to my octoprint gcode start script.

closed 2018-10-20 by KevinOConnor