Klipper3d/klipper #759: Smoothieboard [static_digital_output leds] documentation ?

rtbon, 2018-10-09:
Hello, I'm looking to documentation about the [static_digital_output leds] define in "generic-smoothieboard.cfg"
How are there used in klipper, there is a way to use them as state, like printer error, ready, printing, finish ?
Thanks a lot

pyjamasam, 2018-10-09:
Everything is here: https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/static_digital_output.py
(though you'll have to read the code)

Though that being said, all it does is split the pins and then setup them up as outputs and then turn them on.

If you want to use them similar to the old flasshie mode for smoothie there will have to be some code added.  I could see their value in showing the status of the MCU code and what state things are in.  But again, that's functionality that doesn't exist yet.

chris.

KevinOConnor, 2018-10-11:
The static_digital_output is defined in the example-extras.cfg file - https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

rtbon, 2018-10-11:
Thanks a lot !

closed 2018-10-11 by rtbon