Klipper3d/klipper #760: Fystech TMC2208 Driver

Maximuscr31, 2018-10-09:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2461824/klippy.log)

> Endstop x still triggered after retract

That is the only thing I get. The motor goes firm like it is going to move and has power but nothing actually moves. If I trigger the endstop manually I get the Endstop x still triggered after retract error. I got the Fystech TMC2208 and soldered the 3 pads together and installed X and Y only for testing. I plan to do the other 2 once I am sure it is working.  This is the tutorial I found for soldering the pads on the Fystech 2208. 
I tried switching X and Y drivers to make sure the X wasn't bad. No change. Is this an issue in my config?

Maximuscr31, 2018-10-09:
So I tried increasing current all the way to 1.2 amps in the config. The only thing that happened different was the stepper driver got hot but the motor did not. It just maintained holding torque except for one time it jerked once when I hit home around .9 or 1.0 amps. These are the same motors that came with my Davinci 1.0 and I used them successfully with the A4988 drivers and they still moved with the TMC2208 BigtreeTech but UART didn't work then.

Maximuscr31, 2018-10-09:
Here is a picture 
[image]
of the stepper motor on my X and Y axis. According to this setting :

> sense_resistor: 0.110
>The resistance (in ohms) of the motor sense resistor. The default  is 0.110 ohms.

Should mine be changed to 4.6?

jakep82, 2018-10-09:
Are you sure you have your step distance and microsteps set up correctly?  It might not be the cause of your problem, but a typical step distance would be .0125, not .012371.  You'll probably end up with some weird rounding issues with that step distance.

Also, you shouldn't have microsteps set to 256 at the same time you're using interpolation.  I'm wondering if you actually want 16 microsteps.

Regarding the sense resistor, that's a resistor on the driver and it has nothing to do with the resistance of the motor.

KevinOConnor, 2018-10-09:
I'd start with a default config of just:
```
[tmc2208 stepper_x]
uart_pin: ar44
microsteps: 16
run_current: .9
```

Then run a `FIRMWARE_RESTART`, then run `DUMP_TMC STEPPER=stepper_x`, then home the printer (if necessary, force the x axis to appear as homed), then run `DUMP_TMC STEPPER=stepper_x` again.  Report the results.

-Kevin

Maximuscr31, 2018-10-09:
Seems smooth now. Here is the log file Kevin with the dumps as you requested. The steps is what it came out to from calibration trying to get my 20mm calibration cube well 20mm. I changed it though back to a even number. If all looks well I will install the other two TMC2208 next. Thanks again everyone for the help.

Maximuscr31, 2018-10-10:
I am only now having issues with my extruder. Is this config setting correct?

> [tmc2208 extruder]
uart_pin: analog5
microsteps: 16
run_current: .9

Maximuscr31, 2018-10-10:
Nevermind. The extruder is working with above config settings. The problem was the board had a seat slightly bent at a angle. Where the board is mounted it is hard to notice one side and the pins didn't line up at all on one side. So after I noticed and bent it back and seated it then it worked just fine. Thanks again.

closed 2018-10-10 by Maximuscr31