Klipper3d/klipper #761: Unable to read tmc2208 'extruder' register IFCNT

Mamath2000, 2018-10-09:
Hi
I use FYSETC tmc2208 driver with a FYSETC F6 V1.3 board (it's looks like a classic ramps, but uart for driver are wired).
my klipper installation was old, so i pull the new dev version, but since that, it is impossible to get driver working. 
I always have an error on register IFCNT ! 
TMC2208 works well in standalone mode,  but no way to have them working on uart mode. 

With no hardware configuration, i switch back to marlin (just for test), and no problem, for marlin, uart on driver is working. 

have you an idee of where the problem was?  

thanks, for your help, ...and for your work ! 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2462600/klippy.log)

Mamath2000, 2018-10-10:
Hi, 
i have try, with no succes, the board i used is fully wire. It's mean i have no wire on the top of the driver, the PDN pins is directly plug to the mcu. 
What looks strange is : in the pins.py i can't find the hardware pin used by the mcu for uart/tx . 

for extruder E0, uart is on PJ5 and tx on PJ6 but this two pins are not diffined in pins.py  ? 
Is it normal ?

manoukianv, 2018-10-10:
Hello Math,
you have a FYSETC F6 board and want to used a pin over D69 on a mega2560 mcu.
In pins.py, only pin D0 to D69 are mapped and can be used.

I think you have to add extra pin from D70 to D82 (in the mega2560 pins array) to be able to used them.
Vincent

Mamath2000, 2018-10-10:
Hello Vincent, 
I have try too modify pins.py, by adding extra pin (from D70 to D85). I have still the same error. 

I definitly don't think the pb comes from pins.py.

KevinOConnor, 2018-10-11:
That error indicates that Klipper was unable to access the TMC2208 via the UART.  The uart pin is either incorrect or not routed to the stepper motor driver.

If you only know the Marlin pins then see the FAQ: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name

In particular, D69 is PK7 on the atmega2560.

-Kevin

Gomer05, 2018-10-11:
Having another look at your klippy.log.  I can see that you specify 'pin_map = arduino'. But from the docs the pin names should be 'ar26' etc when using the 'pin_map = arduino', not the 'PC3' names you are using.  Try removing the the pin_map line and see what happens.

I still think there may still be an underlying pin mapping issue but the issue maybe masked by the above config.

As to the pin mappings.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_FYSETC_F6_13.h
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_AVR/fastio_1280.h
Indicates that DIO70_PIN goes to PING4 but we don't see (P)G4 mapped in https://github.com/KevinOConnor/klipper/blob/master/klippy/pins.py
So manoukianv may be right in having to extend the pins.py but not sure if that is all that would need to be done.  i.e. how does the mcu code handle this, does it even care.  

If you try and update the pins.py again.  Post the exact change so we can help validate the change.  

Mark

Gomer05, 2018-10-11:
Question, if using the 'PG4' name which is not mapped in pins.py, do we need to add it to pins.py for 'PG4' to even work?   Or is the mapping just used for those of us who chose to use the 'ar' names?

Mamath2000, 2018-10-11:
> Having another look at your klippy.log. I can see that you specify 'pin_map = arduino'. But from the docs the pin names should be 'ar26' etc when using the 'pin_map = arduino', not the 'PC3' names [...]

Hi Mark, 
Thanks for trying too help me ;) 

I use 'pin_map = arduino' just for the line "sensor_pin: analog12", because i have difficulty to find the good hardware name.  Since my first post, i have found that 'analog12' is the 'PK4' pin. 
So i have try too test without  'pin_map' option, but result is still the same.  

You ask wy i don't use DIO70 pin or PING4, ! the DIO70 correspond to the  "#define X_CS_PIN" that is used in SPI mode for TMC2130. I use TMC2208 so CS_PIN is not used in klipper config. Are you agree ?

 Your last question, is a good question for me !

KevinOConnor, 2018-10-11:
Are you sure Marlin is using them in uart mode (and not in standalone mode)?  I needed to modify my Fysetc 2208 drivers to bridge the solder jumpers.

Also, make sure the motor power is on as the stepper motor drivers are powered by the motor current (and not from the 5V micro-controller power).  They need to be powered up prior to Klipper startup.

Otherwise, I don't have any additional advice - the error indicates that Klipper is unable to communicate with the driver.  If it's the correct pin and that pin is routed to the driver then I'm not sure what the problem is.

-Kevin

Mamath2000, 2018-10-11:
Kevin, 

For marlin yes i'm sure, because drivers have differents microstep. 
M122 command report that 
```
Send: M122
Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	900	900	800	800
Recv: RMS current	887	887	795	795
Recv: MAX current	1251	1251	1121	1121
Recv: Run current	28/31	28/31	25/31	25/31
Recv: Hold current	14/31	14/31	12/31	12/31
Recv: CS actual		0/31	0/31	0/31	0/31
Recv: PWM scale		0	0	0	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	false	false	false	false
Recv: msteps		4	4	4	16
Recv: tstep		0	0	0	0
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:	X = 0x00:00:00:00
Recv: 	Y = 0x00:00:00:00
Recv: 	Z = 0x00:00:00:00
Recv: 	E0 = 0x00:00:00:00
```

For the other point, on power, i have to make somes test. i don't know this information. 
I try to make some more test with a clean install and a very simple printer.cfg. 

I give feedback as soon as possible. 

Thanks for help & time guys
;)

Gomer05, 2018-10-12:
Mamath2000,  This is getting more and more curious.   
I have just done a test with my MKS GEN L board with Marlin 1.1.9.  I'm using one wire as I mentioned before.  
If I have the wire connected to the my TMC2208 on X I get the following result.
>>> M122
SENDING:M122
		X       Enabled		true
...
If I disconnect the UART wire and run M122 again.
>>> M122
SENDING:M122
		X      Enabled		false
...
Note the "Enabled		false" when I disconnect the wire.  
Your post above shows "false" for x, y, z, e0 .  So I think Marlin is not working.

Also if you goto http://marlinfw.org/docs/hardware/tmc_drivers.html
In the troubleshooting section:
"Reported register values of either 0x00000000 or 0xFFFFFFFF are bad responses."
Your output above has the registers: X = 0x00:00:00:00
Where mine with the cable connected was: X = 0xC0:0C:00:00 
and disconnected: X = 0x00:00:00:00

Looking at the Docs for your board can you confirm you have made the needed changes on the board to enable PDN?  It looks like the board is still running standalone, which is the default.

From: https://wiki.fysetc.com/F6_V1.3/
"If you use the TMC2208, there are two modes to choose . FYSETC's 2208 is factory defaulted to standalone mode. At this mode, you can only set the microsteps by MS1 and MS2. If you want to config more , you need to enable PDN. First, You must make the jumper next to the PDN pin closed, and second, make sure that the 2 and 3 pin of the JP1 jumper next to the drive position on F6 board is connected , and then, you can make the configuration of the 2208 with marlin (SoftwareSerial)."

Hope this helps.  

Mark

Gomer05, 2018-10-12:
Mamath2000, one last thing.  I assume you have the TMC2208 V1.0 drivers from FYSTEC.
I have these on the new MKS Gen L board I'm setting up.  In order to get them working in UART, I had to bridge the solder pads on the bottom of the driver near the PDN pin.  
I think thats what the comment I posted above from Fystec is referring to.  "You must make the jumper next to the PDN pin closed"  Have a look at your board and see if it has the solder bridge or not.

Gomer05, 2018-10-12:
As Kevin mentioned in his comment, make sure you have your steppers powered by your 12/24v supply.  I just put klipper back on my board and was able to get the same error you reported.  
I had forgot to power it up by 12/24v....  Once proper power was applied it worked again.

Mamath2000, 2018-10-12:
Hi Gomer05, 

I agree, it's very strange. 
What you say is probably i run tmc in standalone mode. I have to verify, but want you say on the enable parameter and the X = 0x00:00:00:00 are new for me. 

About the PDN bridge, i have solder it. So no pb for that. 

The last point is about my power system, for now the way i powered the board involved that the board is only powered when 12v is on. 
Sorry for my english, but it's difficult to explain! 
I use the 12v line to power the usb cable. in that way when octoprint stop the atx supply, the board is totaly off. 
 Perhaps it's the problem. 

Thank's you for your help.
Bye

Gomer05, 2018-10-12:
Your welcome.

In regards to power, just setup klipper standalone for the moment and make sure that you can command the steppers to move "STEPPER_BUZZ STEPPER=stepper_x" from https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md 
then you'll know that power is good or not.
I've also tried todo a pin mapping as well.  See if that matches yours.
  #define X_SERIAL_RX_PIN  71  PG3
  #define X_SERIAL_TX_PIN  72  PJ2
  #define Y_SERIAL_RX_PIN  73   PJ3
  #define Y_SERIAL_TX_PIN  75   PJ4
  #define Z_SERIAL_RX_PIN  78   PE2
  #define Z_SERIAL_TX_PIN  79   PE6
  #define E0_SERIAL_RX_PIN 76   PJ5
  #define E0_SERIAL_TX_PIN 77   PJ6

Mamath2000, 2018-10-13:
Hi Guy's, 

The issue is close ! There is no bug in klipper. 
@Gomer05 you have probably right, in marlin i'm working in standalone mode. but when i change the microstep of a driver the M122 command report this change but in fact nothing change. 
- finaly my problem is : 2 TMC2208 are "dead", probably not totaly dead, but PDN pin is dead for sure. 

I switch the two driver of X/Y with Z and E0, and set in klipper X and Y setup by UART, and Z and E0 standalone (waiting 2 new TMC). And all work fine. 

Sorry for your time, it was an hardware issue. 
@KevinOConnor thank's for your great work on klipper.  

Bye 
Mamath

closed 2018-10-13 by Mamath2000

khessels, 2020-05-16:
@KevinOConnor  or others, I noticed this bug as well several times on a BTT SKR V1.3, however it always seem to appear random right after i used octoprint to update the printer.cfg. (the fact that i used octoprint might have nothing to do with it i believe) When i waited a while ( i got frustrated and went to bed) It would just disappear, but only if i left the machine off for a while.
I suspect at this point that it might have something to do with the mcu/eeprom not being able to flash correct or something like that when it is to warm ? What did help the last time was to upload a new bin file to the SKR, after which the error disappeared immediately.