Klipper3d/klipper #763: Print Rotation

adamrblack, 2018-10-10:
Hello, loving Klipper on my Anycubic Kossel Linear Plus, but all my prints seem to be rotated 120degs counter clockwise. Delta calibration test printed with A at B tower, B at C tower, and A at B towerfpr example.

Is there a setting I have missed to correct this?

Thanks for the help

KevinOConnor, 2018-10-11:
Sounds like the motor pins are swapped - run the STEPPER_BUZZ command on each stepper (see: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ).  If the motors aren't moving the correct towers then either physically swap the motor cables or swap the motor definitions in the config file.

-Kevin

adamrblack, 2018-10-11:
Thanks Kevin, stepper_buzz moves the correct stepper, I figured out I had to change the angle of each tower and now things are printing the right way around

closed 2018-10-11 by adamrblack