Klipper3d/klipper #764: Idle timeout and resume of the TMC2660 extra

FHeilmann, 2018-10-10:
Hi @KevinOConnor,

when I submitted the TMC2660 extra, we talked about the idle_timeout and resume behavior. The current implementation of the idle_timeout and resume of the extra works good, except during homing moves.

If the homing move takes longer than the idle_timeout, my particular rendition of the idle_timeout fires, because the printer reports a status of "Ready" while homing while in my opinion, it should be "Printing"

Do you have any input on this?

FHeilmann, 2018-10-10:
In order to further test this, I copy pasted the timeout handler from the idle_timeout extra into the TMC2660, to see whether something about my implementation is funky. 

My old code was this:
```Python
    # timeout handler to lower/raise current when entering/leaving the idle state
    # def idle_timeout_handler(self, eventtime):
    #     info = self.toolhead.get_status(eventtime)
    #     status = info['status']
    #     print_time = info['print_time']
    #     if status == 'Printing':
    #         if self.is_idle:
    #             self.set_current(self.current)
    #             self.is_idle = False
    #         return eventtime + self.idle_timeout
    #     estimated_print_time = info['estimated_print_time']
    #     elapsed_time = estimated_print_time - print_time
    #     if elapsed_time < self.idle_timeout:
    #         if self.is_idle:
    #             self.set_current(self.current)
    #             self.is_idle = False
    #         return eventtime + self.idle_timeout - elapsed_time
    #     if not self.is_idle:
    #         self.set_current(float(self.idle_current_percentage) * self.current / 100)
    #         self.is_idle = True
    #     return eventtime + 0.1
```
The new code is this:
```Python
    def run_idle_script(self):
        if not self.is_idle:
            self.set_current(float(self.idle_current_percentage) * self.current / 100)
            self.is_idle = True
    def idle_timeout_handler(self, eventtime):
        info = self.toolhead.get_status(eventtime)
        print_time = info['print_time']
        status = info['status']
        if print_time <= self.last_timeout or status == 'Printing':
            return eventtime + self.idle_timeout
        estimated_print_time = info['estimated_print_time']
        elapsed_time = estimated_print_time - print_time
        if elapsed_time < self.idle_timeout:
            return eventtime + self.idle_timeout - elapsed_time
        self.run_idle_script()
        return eventtime + 1.
```
This code will also lower the current if a homing move takes longer than idle_timeout seconds.

KevinOConnor, 2018-10-11:
Hi Florian,

I think the ultimate solution is to use "events" to signal idle, ready, and printing status as discussed in PR #741.  The idle_timeout code wasn't really designed for ready/printing detection (it was more intended to check if the printer has been totally idle for several minutes).  With a better mechanism for propagating events I think we should be able to more easily close these "races" between ready and printing.

-Kevin

FHeilmann, 2018-10-11:
@KevinOConnor that makes sense. I'm looking forward to the event system ;)

closed 2018-10-11 by FHeilmann