Klipper3d/klipper #765: MCU shutdown due to Rescheduled timer in the past

sk8r776, 2018-10-11:
I am using the same config file that worked perfectly two nights ago and cannot find any reason for the MCU to be shutting down. I tried to lower all print speeds to 10mm/s a try to rule out step issues, and verified the config did not change from the last that worked correctly. I have run out of troubleshooting steps, and any guidance would appreciated on what I'm doing wrong.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2466920/klippy.log)

jakep82, 2018-10-11:
You're moving too fast on the quad gantry level. With your configured step distance, it takes 1600 steps to move all 4 steppers 1mm. At 100mm/s you're at 160,000 steps per second. That's too fast for a MEGA2560, assuming that's what you're using. You need to either reduce your microsteps, or slow your speed on the gantry level.

sk8r776, 2018-10-11:
Wasnt quad_gantry_level, or z_tilt. Changed both to 50mm/s, and it still died right when it started to read the print gcode file.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2467055/klippy.log)

sk8r776, 2018-10-11:
Closing this. This is slicer setting related.

closed 2018-10-11 by sk8r776

pyjamasam, 2018-10-11:
Could you share the slicer settings/issue that caused this?  Just as an FYI for others that might run into the same issue.

chris.

sk8r776, 2018-10-11:
This was a a multitude of issues that caused it, partially slicer related and some config changes were needed.

Slicer had a retraction of 100mm/s, on extruder that was setup for around 560 steps per mm. Outside the capability of mega2560. There was also an error in my config where Z was moving at 100mm/s during my print_start macros.

Broken Config:
```
[gcode_macro print_start]
gcode = 
	G28
	G32
	G0 X350 Y350 F6000
	G0 Z0.2 <-- Z is taking on a feedrate of 100mm/s
	G92 E0
	G1 X300 E10 F3600
	G1 X300 F3600
	G92 E0
	G1 E-3 F3600
	G1 X260 F3600
	G1 Z1
	G92 E0
```

Fixed Config:
```
[gcode_macro print_start]
gcode = 
	G28
	G32
	G0 X350 Y350 F3600
	G0 Z0.2 F3600
	G92 E0
	G1 X300 E10 F3600
	G1 X300 F3600
	G92 E0
	G1 E-3 F3600
	G1 X260 F3600
	G1 Z1
	G92 E0
```