Klipper3d/klipper #770: [FR] Wiki

luxarts, 2018-10-11:
Hello! I have to congratulate you for this awesome job.

I was looking in the repository reading the docs and noticed that you have no wiki so why not enable it? We can add the documentation there and will be more readable than reading directly from plain files.

Another question is why is Octoprint needed? Would be good an standalone firmware. I would be glad to help you as I'm developing something similar but using an ESP12F (MCU with WiFI) in the same way that you are using the Raspberry Pi.

Regards,
Lucas

jakep82, 2018-10-11:
It already is a standalone firmware from an operational standpoint.  You can do pretty much everything through a standard RepRap LCD, and  Repetier Server also supports Klipper now.

KevinOConnor, 2018-10-13:
> I was looking in the repository reading the docs and noticed that you have no wiki so why not enable it? 

I have not had a positive experience with any wikis for technical projects.  (Wikipedia itself is great, but I've not seen it replicated anywhere else.)  In my experience, wikis rapidly become stale, contradictory, and cluttered.

It's possible to add documentation to Klipper by submitting a pull request.  The markdown format of the documents should render okay when viewed from the github site.  I know this isn't a great solution, but I don't think the alternatives are an improvement.

-Kevin

Avalonnw, 2018-10-16:
Something like readthedocs.io is a great looking git wiki thing, but pretty complicated IMO.

lf-, 2018-10-28:
It looks like the GitHub wiki functionality is basically an auto-approved pull request with fewer features than one, which is almost what we have already with documentation changes, only ours are reviewed ;)

closed 2018-11-07 by KevinOConnor