Klipper3d/klipper #772: Error flashing firmware in Arduino DUE

luxarts, 2018-10-12:
Hello! I followed all the steps in the installation guide and when I execute `make flash FLASH_DEVICE=/dev/ttyACM0` it start to flash but at the end throws `src/sam3x8e/Makefile:38: recipe for target 'flash' failed`.

The complete log:
```
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/ttyACM0
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/sam3x8e/main.o
  Compiling out/src/sam3x8e/timer.o
  Compiling out/src/sam3x8e/gpio.o
  Compiling out/src/sam3x8e/spi.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/../lib/sam3x/gcc/system_sam3xa.o
  Compiling out/src/../lib/sam3x/gcc/gcc/startup_sam3xa.o
  Compiling out/src/sam3x8e/serial.o
  Compiling out/src/generic/serial_irq.o
  Building out/compile_time_request.o
Version: v0.6.0-529-gaa693fb-20181012_052643-octopi
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
  Flashing out/klipper.bin lib/bossac/bin/bossac to /dev/ttyACM0 via bossac
Erase flash

Done in 0.061 seconds
Write 15152 bytes to flash (60 pages)
[==============================] 100% (60/60 pages)
Done in 3.663 seconds
Verify 15152 bytes of flash
[==============================] 100% (60/60 pages)
Verify successful
Done in 3.158 seconds
Set boot flash true

SAM-BA operation failed
src/sam3x8e/Makefile:38: recipe for target 'flash' failed
make: *** [flash] Error 1
```

I tried uploading a test code to the same board with Arduino IDE and it works perfectly (the upload an the program). Is there a way to upload the firmware with Arduino IDE?

Thanks!

KevinOConnor, 2018-10-13:
That's an error reported by the "bossac" flash writing application.  I've seen it do that a few times as well and I'm not sure why.  In my experience it does actually flash the device, but you can rerun the "make flash" command again to be sure.

-Kevin

closed 2018-10-13 by luxarts

KevinOConnor, 2018-10-22:
It looks like the error report is due to the post flash restart request - not a failure during the actual flashing.  It also appears it is a spuious error.  I've updated the build to suppress it (commit 1a437c1f).

-Kevin