Klipper3d/klipper #775: Delta_calibrate position_endstop off by 2x z_offset

b0ssman, 2018-10-12:
My Position endstop is about 298

This is my Probe section

[probe]
pin: ^ar18
x_offset: 0
y_offset: 0
z_offset: -15.9

When i do delta_calibrate i get these values, but the position endstop should is now off by 2x z_offset
#*# [stepper_a]
#*# angle = 209.645809
#*# arm_length = 267.000000
#*# position_endstop = 264.696399
#*#
#*# [stepper_b]
#*# angle = 329.974449
#*# arm_length = 267.000000
#*# position_endstop = 266.061117
#*#
#*# [stepper_c]
#*# angle = 89.881535
#*# arm_length = 267.000000
#*# position_endstop = 266.662992
#*#
#*#
#*# [delta_calibrate]
#*# height0 = -15.9
#*# height0_pos = 22536.000,22536.000,22536.000
#*# height1 = -15.9
#*# height1_pos = 27952.667,27986.667,19852.667
#*# height2 = -15.9
#*# height2_pos = 22002.333,31205.333,22033.333
#*# height3 = -15.9
#*# height3_pos = 19816.667,26925.667,26934.667
#*# height4 = -15.9
#*# height4_pos = 21814.000,21798.000,28234.000
#*# height5 = -15.9
#*# height5_pos = 26221.667,20043.667,26188.667
#*# height6 = -15.9
#*# height6_pos = 29669.333,21969.333,21963.333

The Printer when moved to Z0 afterwards now sits at 2x offset

BlackStump, 2018-10-13:
I think your offset should be 15.9 not -15.9 
a offset of -15.9 would mean your probe triggers 15.9mm below the surface of your bed

eg
#z_offset:
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided

KevinOConnor, 2018-10-14:
You'll need to attach the klipper log file in order for us to assist - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

b0ssman, 2018-10-16:
i switched back to marlin.
if it is plus then its confusing coming from marlin the z offset is negative

closed 2018-10-16 by b0ssman