Klipper3d/klipper #777: Klipper on Cetus3D - can't get Z to zero reliably

yoyo42, 2018-10-13:
(Logfile at end)

I'm setting up Klipper to work with my Cetus3D fitted with a Tinyfab (smoothie) controller. I'm nearly there, it's printing reasonably well and quickly. The remaining big issue is that I can't get Z homing to work, the zero position varies every time I do large Z moves (to the Zmax microswitch endstop or not) by up to 20mm. But when it's printing the layer heights seem to be OK.

I'm driving the Z axis at approximately the same slow speed (8mm/s) as the original Cetus software used. X and Y I have driven up to 250mm/s and they seem to work fine and repeatably. My Z axis is counterbalanced using a spring rewinder (like a tape measure coil spring) so is effectively much lighter than the standard setup, putting less load on the stepper. However I only did that because the standard setup, with a friction brake to avoid the arm dropping when the steppers were de-energised, was showing the same issue.

I can't see why the Z won't zero reliably. When I was using it with Smoothieware I set the bed height once and used it for a couple of months with no need to ever change it by even 0.1mm. Any ideas or things to test would be welcomed...

Thanks in advance

The attached log file was taken after the following test:

- load model into Octoprint
- start printing
- pause at Z=50 while the hotend heats
- use Octoprint controls to do the paper height test
- M114 shows Z5.3
- set OctoKlipper 'Coordinate offset' to Z=5.3 (so Z=0 is now the bed height)
- use the terminal to move a lot:
- G1 Z150
- approach in steps to avoid a bed crash, ended with G1 Z0
- paper is dragging but a little looser, so almost OK
- rehome: G28 Z
- find bed height again using G1 Zxx moves (5.3mm offset still in place)
- bed height is now Z=11.4
- M112
- attach log

[zhome-issue.log](https://github.com/KevinOConnor/klipper/files/2475291/zhome-issue.log)

KevinOConnor, 2018-10-14:
It's not clear if you are reporting that the z stepper is unable to keep its position, or if you are reporting that a 'coordinate offset set in OctoKlipper' is getting lost.

I recommend you tune the Z position_endstop so that no runtime offset is needed.  If you lose steps after that, then I'd look at stepper current and/or reduce stepper speed.

I also recommend you update to the latest code.  It appears you've modified Klipper - that will make it harder for us to assist.  In particular, statistics are no longer making it to the log.

-Kevin

yoyo42, 2018-10-14:
Thanks, I have now updated with a git pull and it acts exactly the same. The only mods I made were to work with a different temperature probe, and in the process I uncommented some of the debug code.

I can adjust the Z endstop position in printer.cfg for a zero offset but it will be wrong again after the next G28 homing cycle.

The stepper current on this printer is set in hardware, and worked well with Cetus or Smoothie software. I've reduced the speed and acceleration down to 1 and 1 and it did the same thing, but it wasn't as far out. It's at 8 and 50 now which appears to be very close to Youtube videos of the original speed.

I'm starting to think there must be something mechanically wrong with it - I think the only way to test that will be to put Smoothie back on it.

KevinOConnor, 2018-10-23:
Okay - I'm going to close this for now.  I'll reopen if you need further assistance.

-Kevin

closed 2018-10-23 by KevinOConnor

dharvey242, 2019-10-15:
Is it possible to share the config you created for the Cetus and tinyfab CPU?