Klipper3d/klipper #779: Can not use SAVE_CONFIG

blaskkaffe, 2018-10-13:
Is the SAVE_CONFIG command a part of the current master branch? Tried to do a git pull and update to the latest but the icons are the same as before (says fan instead of the icon that waa merged to master) and there is no way to save the delta probing to cofig, except for copying and pasting. Am I missing something?

cosminr86, 2018-10-14:
Have you restart klipper service?

blaskkaffe, 2018-10-14:
Yes, i have restarted klipper service, reflashed the controller muktiple times. Tried the same update procedures multiple times and when I check it says I have the latest version of master branch.

To me it seems like the flashing doesnt work even tho everything looks successful.

KevinOConnor, 2018-10-14:
You'll need to attach the klipper log file in order for us to assist - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

blaskkaffe, 2018-10-14:
When I stopped the klipper service AND powercycled the MCU so that it didnt boot it worked after rebuilding and flashing. 

Still not getting the new icon even tho the local files looks allright. Will take a look at the log and send here.

KevinOConnor, 2018-10-23:
Okay - let us know if you are still having the issue.

-Kevin

closed 2018-10-23 by KevinOConnor