Klipper3d/klipper #783: No option 'sensor_type' in section: 'heater_bed'

rustam-rezvanov, 2018-10-15:
Hello!
first time I install a clipper
and get this error

Send: status
Recv:
// Klipper state: Not ready
Recv:
// No option 'sensor_type' in section: 'heater_bed'
Recv:
// Once the underlying issue is corrected, use the "RESTART"
Recv:
// command to reload the config and restart the host software.
Recv:
// Printer is haltedRecv: !! No option 'sensor_type' in section: 'heater_an

in the configuration file, this section looks like this:

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: pid
pid_Kp: 332.15
pid_Ki: 46.07
pid_Kd: 598.62
min_temp: 0
max_temp: 110

Help me to understand what is the reason?

cosminr86, 2018-10-15:
Post klippy.log

KevinOConnor, 2018-10-16:
As mentioned, in order to assist we need the klipper log file.  See: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

closed 2018-10-22 by KevinOConnor