Klipper3d/klipper #784: avr-gcc environment octopi 0.15.4

VincentGijsen, 2018-10-15:
the installation of octopi is assumed to contain the proper build files and headers, but my experience is different on octopi 0.15.4,

I had to install the following, to complete the installation.md

version:
```
OctoPrint version : 1.3.9
OctoPi version    : 0.15.1
```
issue after avr-gcc: `src/sched.c:7:30: fatal error: setjmp.h: No such file or directory
`
```
sudo apt-get install libncurses5-dev
sudo apt-get install avr-gcc
sudo apt-get install avr-libc
```

perhaps extend the installation.md to contain the above (as a single command).

VincentGijsen, 2018-10-15:
edit: nevermind, i didn't run the install script...

closed 2018-10-15 by VincentGijsen

loktar00, 2021-07-13:
Had the same error, doing this on WSL so I glossed over the octopi stuff. For anyone who doesn't know what the install script is, it's referring to the ones located in `./klipper/scripts/` I had to run the one for Ubuntu just as an example.