Klipper3d/klipper #785: Timeout on serial connect

andresgoku, 2018-10-15:
I was following the tutorial on youtube how to install and connect Klipper but after flash the printer board i can get the status command on octoprint.

This is in octoprint 

Send: status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok

and the the KliPPY.LOG 

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-536-g21597f9'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'
Start printer at Mon Oct 15 12:07:12 2018 (1539605232.8 9.2)
===== Config file =====
[stepper_x]
step_pin = ar54
dir_pin = !ar55
enable_pin = !ar38
step_distance = .0125
endstop_pin = ^ar3
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = .0125
endstop_pin = ^ar14
position_endstop = 0
position_max = 200
homing_speed = 50

[stepper_z]
step_pin = ar46
dir_pin = ar48
enable_pin = !ar62
step_distance = .0025
endstop_pin = ^ar18
position_endstop = 0.5
position_max = 250

[extruder]
step_pin = ar26
dir_pin = ar28
enable_pin = !ar24
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = ar10
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog13
control = pid
pid_kp = 21.556
pid_ki = 1.065
pid_kd = 109.128
min_temp = 0
max_temp = 250
max_extrude_cross_section = 50.0
max_extrude_only_distance = 500.0
max_extrude_only_velocity = 120.0
max_extrude_only_accel = 1250.0
pressure_advance = 0
pressure_advance_lookahead_time = 0.1

[heater_bed]
heater_pin = ar8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = analog14
control = pid
pid_kp = 52.642
pid_ki = 0.742
pid_kd = 933.741
min_temp = 0
max_temp = 130

[fan]
pin = ar9

[heater_fan nozzle_fan]
pin = ar7
heater = extruder
heater_temp = 50.0
fan_speed = 1

[mcu]
serial = /dev/ttyUSB0
pin_map = arduino

[printer]
kinematics = cartesian
max_velocity = 250
max_accel = 3000
max_z_velocity = 10
max_z_accel = 100
=======================
Extruder max_extrude_ratio=20.787584
Starting serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Got error 0 in read: (0)Success
Timeout on serial connect
Unable to open port: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.
Klipper state: Not ready
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect
Timeout on serial connect

[klippy.log](https://github.com/KevinOConnor/klipper/files/2478787/klippy.log)

thanks for the help 

and I already read mostly all the issues closed

BlackStump, 2018-10-15:
did you look at the docs? there is a wealth of info in them, try 
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

KevinOConnor, 2018-10-16:
As mentioned, this is usually the result of the micro-controller not being flashed, not being on, not having the correct baud rate (make sure to compile the micro-controller is compiled with a baud of 250000), or not having the correct port definition.  See the FAQ for solutions.

andresgoku, 2018-11-01:
Kevin I check the baudrates to be sure of the right number and i no hable to get "printer ready"

[image]
[image]
[image]
[image]
[image]
[image]

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2538462/klippy.zip)
[printer.zip](https://github.com/KevinOConnor/klipper/files/2538463/printer.zip)

Printer ready"

jakep82, 2018-11-01:
Why do you have the processor speed configured for 8MHz?  A standard 2560 is 16MHz.

jakep82, 2018-11-01:
Also, the Ender 3 has an atmega1284p chip.  Unless you've changed the control board, you compiled for the wrong processor.

BlackStump, 2018-11-01:
The Ender 3 also does not have a bootloader from factory so that might be the first hurdle to jump ....plenty of info on the net on what's required to flash a bootloader to that board and a quick search found some youtube vids .

andresgoku, 2018-11-01:
Sorry I forgot to add than the board is a MKS Gen L
But with information of Jakep82 I change the processor speed to 16 MHz

Thanks so much guys

closed 2018-11-01 by andresgoku