Klipper3d/klipper #786: Unable to parse option 'z_offset' in section 'probe'

oleplay, 2018-10-15:
When trying to configure Klipper, it is "Unable to parse option 'z_offset' in section 'probe'". 

[klippy.log](https://github.com/KevinOConnor/klipper/files/2479809/klippy.log)

Thanks for the help

mwdeez, 2018-10-16:
My [probe] section (and most of my printer.cfg) is configuring using the `key: value` notation instead of `key = value`. Since the error presents with any `*_offset` that may help.

Here's the excerpt from my config:
```

KevinOConnor, 2018-10-16:
The Python config parser doesn't like the indented comment following z_offset.  You can either remove that comment, move the comment so it is at the start of the line, or upgrade to the latest software (with commit 08d03ae0).

-Kevin

oleplay, 2018-10-16:
Thanks for the quick fix, updating fixed the issue

closed 2018-10-16 by oleplay