Klipper3d/klipper #789: Change config settings "on the fly" - how hard would it be?

swatti87, 2018-10-17:
Been trying  to get settings in order but re-starting the print every time is kinda slow.

How hard would it be to make it possible to adjust these settings via a console-command "on the fly" ? 

I have no coding experience or much of understanding about this but the thought came into my mind after starting like the ~30th test-print.

cosminr86, 2018-10-18:
you can config a lot of settings on the fly. just read the docs (including g-code.md)
what do you try to do ?

swatti87, 2018-10-19:
Im mainly hoping to configure "pressure advance" on the fly without having to re-start the print.
Can i do that?

jakep82, 2018-10-19:
The documents have all the answers.

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

swatti87, 2018-10-19:
> The documents have all the answers.
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

Oh goodie. Must have missed the PA setting there. Looks like all the stuff i want to change are there. 

Closing.

closed 2018-10-19 by swatti87