Klipper3d/klipper #794: Motors and fans enables when RPI disconnect

ffnull, 2018-10-18:
When im shutdown RPI from power my motors and funs go to state "ON".
How to fix this?
Board RUMBA
Raspberry Pi3

megablue, 2018-10-19:
Don't disconnect.

KevinOConnor, 2018-10-20:
You'll need to attach the klipper log file for us to assist - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Also, be sure to run the config checks described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md

ffnull, 2018-10-20:
@KevinOConnor what log? RaspberryPi is off. so there problem on microcontroller firmware

megablue, 2018-10-22:
@ffnull with Klipper you are not supposed to treat raspberry pi as a host computer but part of the 3d printer firmware itself, so you are not supposed to power off the raspberry pi or disconnect it from the printer while the unit as a whole is powered on.

blaskkaffe, 2018-10-23:
I have noticed the same thing.

Thought it was a safety feature to run the fans at full speed if the raspbeeey pi disconnects since it is half of the firmware, the part that does most of the work. Would be bad if it just continued heating. This is most likely a fail safe feature.

KevinOConnor, 2018-10-27:
Without a log, there is little we can do to assist.  Only advice I can provide is to run through the config checks described at https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md - in particular make sure the micro-controller does the correct thing during an M112 command.  If the micro-controller does set the correct pin state during an M112 but does not when power is removed from the board's USB port, then it's likely that the micro-controller is powered over USB and the hardware is defaulting some pins on - that's a hardware issue that can't be addressed in software.

-Kevin

closed 2018-11-07 by KevinOConnor