Klipper3d/klipper #797: [FR] SET_NOZZLE_SIZE

Hywelmartin, 2018-10-19:
I haven't found any way(other than edit printer.cfg) to alter nozzle size

So my suggestion is to have a extended gcode like:
 SET_NOZZLE_SIZE 0.4
that could be a part of start gcode or just punched in in the octopi terminal

And by the way GREAT FW...

/Martin

FHeilmann, 2018-10-19:
Nozzle size is not something the printer is aware of. This is entirely handled by the slicer. Which functionality do you expect to achieve with this command?

FHeilmann, 2018-10-19:
It's used for a security feature where the firmware rejects moves that extrude more material than the hotend can handle. Unless you are running into issues where your printer rejects gcode due to too much filament extruded you can leave this value at 0.4 and simply ignore it.

Hywelmartin, 2018-10-19:
confusing..
The volumetric flow is not dependent on the nozzle size.... 

To be able to know the volumetric flow you need the steps/mm from the extruder and the filament diameter... not the nozzle size... what you put in shall get out... if the fw depend on the nozzle size to determent the volumetric flow there is at least one calculation to many...

I switch nozzles rather frequently  so I taught that it would be beneficial to be able to adjusting the parameters in another way... So I can put at a arbitrary value and forget it.... good to know..

/Martin

Hywelmartin, 2018-10-19:
well I'm not a coder so ... i don't get anything out of this... well something

but my point stands if it's a useful parameter that we would be able to adjust it by extended gcode..

feel free to evaluate and take the next step..

jakep82, 2018-10-19:
As @FHeilmann noted this is only a safety feature.  If you're going to change nozzles, it's not unreasonable to also update your config, so this doesn't need to be run-time configurable.  I just leave mine at 0.5 which is the largest nozzle I ever use, and don't bother changing it.

faboaic, 2018-10-24:
I often print with 0.4mm nozzle diameter but extrusion width 0.8 - 1.2 mm. 
So this safety feature just ruined a print of mine. 

Now I set nozzle diameter in printer.cfg to 0.8mm and see what happens...

FHeilmann, 2018-10-24:
your use case is what the `max_extrude_cross_section` configuration parameter is for. By default it is `4*nozzle diameter^2` which allows the equivalent of .8 extrusion width for a .4mm nozzle.

In your case (with a 1.2mm extrusion width) you can set this to 4 * 0.6^2 (=1.44) to raise the safe limit and allow you to extrude up to 1.2mm extrusion width with a 0.4mm nozzle.

faboaic, 2018-10-24:
I will try that setting. 

Klipper itself did not ruin the print but the default configuration is too conservative regarding extrusion width. 
A factor of 2.5 or 3 is no problem for most scenarios. 
I know... Many users think that with 0.4mm nozzle you can only print 0.4 - 0.48mm extrusion width.  ;)

What really would help :
Run a sanity check or simulation of the whole gcode before the print instead of cancelling it after some hours...    

Thanks,  Fabian

KevinOConnor, 2018-10-24:
Just to be clear, the max_extrude_cross_section specifies an area - it is the extrusion height multiplied by the extrusion width.  With a layer height of .3 and a nozzle diameter of .4, the default value would permit an extrusion width of 2.133mm.  The default value is significantly larger than any sane printing value.

-Kevin

Hywelmartin, 2018-10-25:
thank for your time...

Hywelmartin, 2019-07-06:
@KevinOConnor 
Just for clarification...

The nozzle_size or any derivatives of set size (like "max_extrude_ratio"), does it have any other function than to give an error.
Does it have anything to do with PA??
I'm asking due to the fact that it has been claimed that there is an impact on PA...

closed 2019-07-06 by Hywelmartin

ardyesp, 2021-02-10:
Here is another vote for this feature.

Changing the Klipper configuration, every time nozzle is changed does not work well in a workflow. It is much streamlined to have a ```SET_NOZZLE_SIZE``` gcode.
In the absence of this gcode, we are forced to configure an arbitrarily large value for ```max_extrude_ratio``` and loose the over-extrude protection offered by it.

BTW: Excellent firmware, I love it!

tarsonis123, 2021-02-27:
So this looks like this feature won't see any light, what is the max value I have to put in max_extrude_cross_section to use a 1.0 nozzle with default ext. width and don't need to set this value ever again swapping to smaller sizes afterwards? I'm dumb af and have no clue about the math cheers

clinta, 2021-03-28:
It's a shame this is closed. This would be really helpful because I could configure prusa slicer to use the `nozzle_diameter` variable to correctly set the nozzle diameter in klipper to match what the slicer is configured for at the start of every print.

I've started doing this for pressure advance, using `SET_PRESSURE_ADVANCE ADVANCE` as custom gcode in the filament settings to have correct pressure advance for each filament. I'd like to have all the settings that I might be changing on a regular basis be able to be set by the slicer.