Klipper3d/klipper #799: klipper and machinekit

mngr0, 2018-10-19:
Hi,
I am thinking about a little project, and I would like to hear the opinion of some experienced klipper user.

I would like to "port" the software you have made for the MCUs for machinekit.
Machinekit uses HAL modules, and it's possible to make one that receives the commanded position from the trajectory planner and send that (or the commanded velocity) to the MCU.
This will allow to use all the machinekit hal stack, with all its features, and without being forced in the hardware selection.

Various attempts in this direction have already been done (  [PiCNC](https://github.com/kinsamanka/PICnc-V2), [hal2arduino](https://openbuilds.com/projectresources/hal2arduino.124/updates)  ), but they are no longer mantained.

What do you think about it? 
(If you have a forum or a more suitable place for this topic I will move there)

I have read your protocol.md and mcu_commands.md and the protocol seems clear, but I had trouble in finding an implementation of it, particularly on the MCU side

KevinOConnor, 2018-10-20:
The mailing list may be a better forum - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

A couple of years ago (prior to my starting the Klipper project) I took a brief look at Machinekit.  The impression that I got was that its HAL modules worked using the concept of "perform action X now".  In contrast, Klipper works on the concept of "perform action X at time T".  That is, the Klipper host software generates a schedule of actions that should be performed in the near future, and then the Klipper mcu software executes that schedule.

There are some big advantages to using a schedule - it allows for more precise timing, it allows the host to compress the schedule for easier transmission and storage, it makes execution of the schedule more cpu efficient in the micro-controller, etc.  There is also a cost - it does require additional software complexity - in particular the host software that generates the schedule must be aware of the different "time lines" that actions occur at.

I doubt the klipper mcu code would work well without getting a schedule of events, and I suspect it would be a great deal of work to change Machinekit to generate a schedule of events.

> I have read your protocol.md and mcu_commands.md and the protocol seems clear, but I had trouble in finding an implementation of it, particularly on the MCU side

The https://github.com/KevinOConnor/klipper/blob/master/docs/Code_Overview.md document has some pointers on code location and code flow.

-Kevin

mngr0, 2018-11-11:
The HAL layer has 2 type of interface:
the one you that saw between the modules (represented with pin and signals);
and buffers for low level code (called ring buffers). This are used to communicate with the real time environment (not the MCU).
Look [this image](https://machinekoder.com/wp-content/uploads/2016/11/scope-1.png) taken [from here](https://machinekoder.com/machinetalk-explained-part-1-introduction/):
You can see the Gcode interpreter, the Task executor and the UI talking with each other using HAL (actually in this image they are using MachineTalk, in few words, HAL with network capabilities); and using ring buffers to talk with the planner and the motion controller.
In the BeagleBone the controller is the one that sends the commands to the PRU (directly), it should be enough to modify it in order to use klipper protocol with MCUs.

more on Machinekit: [video1](https://www.youtube.com/watch?v=Bg38894LPPg) [slides1](http://static.mah.priv.at/public/gcode-interpreter.pdf) [video2](https://www.youtube.com/watch?v=xYRNG1Yi1AM) [slides2](http://static.mah.priv.at/public/motion.pdf)
and here he says that the motion controller is a old mess of code, but maybe it's just enough replace it with something from klipper.

At the moment the machinekit community is splitting its software, they are dividing the HAL from the CNC part (you can see two different repositories [machinekit-hal](https://github.com/machinekit/machinekit-hal) and [machienkit-cnc](https://github.com/machinekit/machinekit-cnc)).

That said, I apologies for the late reply, I needed a few days to study some thing here and there.
You did well closing this, because it is not really an issue.

KevinOConnor, 2018-11-14:
Thanks.  Very interesting.  I do see a lot of overlap between Klipper and Machinekit (gcode parsing, lookahead, motion planning done in cartesian space, kinematics done separately, forward/reverse kinematics, etc.).

However, it still seems as if Machinekit is designed around the idea of "figure out what needs to be done" and then "try to execute those actions as quickly as possible".  Whereas Klipper fundamentally works on the idea of "figure out what needs to be done at what times" and then "schedule those events to be run at precisely the requested times".

I suppose one could pair the Machinekit high level logic with the Klipper motion planner driving the Klipper micro-controller code.  (Thus using Klipper's time scheduling motion planner.)

It is interesting and it would be good to extend Klipper to CNC machines.  Let me know what you find.

-Kevin

KevinOConnor, 2018-11-30:
> But my concern is about things like encoders, how could klipper manage them?
The couple encoder+motor is very common and it might deserve a dedicated piece of code in the MCU firmware. But what for a different use of encoders?
>
> A "similar" problem is probing, moving an axis until something happen, like homing.
I looked how klipper manages homing, correct me if I am wrong, it moves little by little until the switch is pressed.
> 
> More in general, any queue-buster event that could interrupt or modify the event scheduled for the MCU.

For these types of "queue buster" situations, we generally add code to the micro-controller so that it can respond to the event with low latency.  An example of this is the homing/endstop code - during homing, the host sends the necessary homing movement to the micro-controller (using the regular kinematic movement system) and instructs the micro-controller endstop code to halt a set of steppers when the endstop switch triggers.  So, the endstop code in the micro-controller knows how to handle that particular "queue busting" event.  Another example is the micro-controller thermocouple code - it is configured to regularly query the thermocouple chips and it is configured what the valid temperature range is.  Should the temperature exceed the configured range then the thermocouple code knows how to shutdown the heaters even without host intervention.

In contrast, when external events don't require a fast response, we typically perform the logic in the host code.  An example of this is the PID temperature control loop.  In this situation, the mirco-controller is configured to take measurements every 300ms and forward them to the host.  The host uses that info to calculate updated heating parameters and schedules the corresponding heating pin changes to take effect just after the next measurement (that is ~300ms in the future).

I suspect motor encoders would follow a similar mechanism - code in the micro-controller to respond to encoder feedback quickly along with code in the host to configure and monitor it.

> Klipper does not use a real time kernel, and the protocol to send commands to the MCU includes the clock time of execution.
did you add the clock time because of the lack of real time kernel?
If Klipper started using a rt kernel (somehow), would be possible to simplify the serial protocol?
In particular would it be possible to send them without the time clock?

The scheduling of events with precise timing is a key feature of Klipper.  It was an initial design goal and it is working well.  That capability is key to Klipper's ability to utilize low cost micro-controllers, deploy multiple micro-controller as a single logical unit, obtain high performance, and have portable code across both micro-controllers and host hardware.

Utilizing a real-time kernel helps to reduce latency in software running on general purpose computers, but it would not reduce the latency/jitter associated with transmission of commands to a micro-controller.  Thus a real-time kernel would not provide any particular benefit in the Klipper architecture.  (That is, a real-time kernel only really helps if everything - including the motion planner - runs as a real-time process on a single cpu.)

-Kevin