Klipper3d/klipper #801: VellemanK8200/3Drag pin config

PerLangkjaer, 2018-10-20:
Hello

I have now been messing around with the config file to get klipper working with my K8200 but with no luck.

Is it possible that someone Cold help me with an exaple config file with the correct pinout?

best regards

dizzey1, 2018-10-21:
From the FAQ
--------------------------------------------------------------------
How do I convert a Marlin pin number to a Klipper pin name?
Short answer: In some cases one can use Klipper's pin_map: arduino feature. Otherwise, for "digital" pins, one method is to search for the requested pin in Marlin's fastio header files. The Atmega2560 and Atmega1280 chips use fastio_1280.h, while the Atmega644p and Atmega1284p chips use fastio_644.h. For example, if you are looking to translate Marlin's digital pin number 23 on an atmega2560 then one could find the following line in Marlin's fastio_1280.h file:

#define DIO23_PIN PINA1
The DIO23 indicates the line is for Marlin's pin 23 and the PINA1 indicates the pin uses the hardware name of PA1. Klipper uses the hardware names (eg, PA1).
----------------------------------------------------------------

So your best bet would be to try to backtrace it from the marlin config files since marlin supports the k8200

dizzey1, 2018-10-29:
Great you should se if you can get your config added as a sample so it is eassier for the next one