Klipper3d/klipper #802: Stuttering on CR-10S5

FixFuture, 2018-10-20:
I recently moved from marlin to klipper and i love the concept. I have managed to configure bltouch and titan extruder successfully(?).

Few issues that i would like to get some help:

For some reason printer stutters (stops briefly and continues) even on straight line. Any ideas what could cause this?

I also get 'Move out of range' error. I fixed it by adding 'position_min: -1' to config.

[Video of stutterging movement](https://photos.app.goo.gl/cCUMkhduHcZ7jiCg8)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2498025/klippy.log)

Arksine, 2018-10-20:
I think that this is due to a combination of mesh leveling and square corner velocity being set to zero.  Bed_mesh is slicing up moves when a split is detected to make a z adjustment, and when square corner velocity is set to zero junction_deviation is also set to zero.  I tested this myself (using kinematic moves only) and noticed stuttering when SQV is set to 0.  Works fine when its set to the default of 5.

Try a print without setting square corner velocity to 0 and see if it still stutters.  If you are setting it to zero to calibrate pressure advance, you can set 'move_check_distance' in  [bed_mesh] to a higher number to prevent smaller splits, such as 30 or 50.  Just be aware that this could affect layer adhesion on large prints, so I suggest setting it back so default by removing the option, or setting it to a relatively sane value like 10 or 15 (assuming you aren't using an SQV of 0)

Also, it is common for mesh leveling to move the tool below zero.  This is normal behavior if parts of the bed are lower than the homed position.  It is correct to set position_min to a negative number when using mesh leveling.

FixFuture, 2018-10-21:
Removing corner velocity setting helped.

Thank you!

So if i want to configure pressure advance, i should add corner_radius and move_check_distance? What does look ahead do?

KevinOConnor, 2018-10-25:
FYI, I've updated the documentation so that the pressure advance tuning guide no longer recommends setting square_corner_velocity=0 (commit ca7dd8bf).

-Kevin