Klipper3d/klipper #803: slic3r pe causing mcu shutdown

tony978, 2018-10-20:
so every time I run code from that program I get a mcu shut down , all the m900/907 are removed . It will heat bed/ hotend to temp , then home and extruder motor moves a little then crash . No probe bed leveling has been done . 
The gcode is attached 
[cali_1-R.txt](https://github.com/KevinOConnor/klipper/files/2498605/cali_1-R.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2498599/klippy.log)

Hywelmartin, 2018-10-21:
G1 E35.6000 F180 -> Extruder max_extrude_ratio=187.088260   ?

try to change the speed or length of that extrusion

tony978, 2018-10-21:
I'm using a titan geared extruder .

tony978, 2018-10-21:
@Hywelmartin issue was pressure advance and fast moves .

mamoit, 2018-10-26:
@tony978 So this can be closed?

tony978, 2018-10-26:
@mamoit should still be something that's needs to be addressed in the code to deal with max stepper moves and pressure advance.  Note I'm  using a geared extruder.

KevinOConnor, 2018-10-27:
There is a check - it's max_extrude_cross_section - unfortunately your config disables it.

-Kevin

tony978, 2018-10-29:
@KevinOConnor  thank you! I will enable it

tony978, 2018-10-30:
[klippy (1).log](https://github.com/KevinOConnor/klipper/files/2531750/klippy.1.log)

@KevinOConnor  I just got a chance to change the config . max_extrude_cross_section was enabled , I lowered the values from 450 to 50 and go the same mcu error . log and gcode are attached . 
[test-gcode.txt](https://github.com/KevinOConnor/klipper/files/2531767/test-gcode.txt)

KevinOConnor, 2018-10-31:
A max_extrude_cross_section=50 would mean it's okay for your .4mm nozzle to extrude a line that is 7mm by 7mm - it's an absurdly large value.  The default max_extrude_cross_section for a .4mm nozzle is 0.640 .

If you return the max_extrude_cross_section to its default value you will no longer get the above error - instead you'll get a "Move exceeds maximum extrusion" error.  The troubling g-code is:
```
G1 X5.000 Y0.260 F7200
G1 X64.625 E82.6000 F2072
```
That last line requests an X move of 59.625mm while extruding 82.6mm in filament.  (At a layer height of .3mm it would be a request for an extrusion width of 4.6mm.)  That does not seem like a valid request.

When the max_extrude_cross_section check is disabled (by setting it to an absurdly large value) klipper will permit the request, but to build up the necessary pressure for it (with pressure_advance=.3) requires moving the extruder at a step rate faster than your micro-controller can achieve.

My suggestion is to use the default max_extrude_cross_section and remove the questionable g-code.  If this is from some kind of extruder priming sequence, then it should be possible to rework it without requiring some high extrusion rates.

-Kevin

tony978, 2018-10-31:
That line is loading the filament from the parked position and not extruding plastic . Can I activate/deactivate pressure_advance from gcode during the tool change ?  This is only an issue when using slic3r pe  .

KevinOConnor, 2018-11-07:
Yes - it's possible to change pressure advance via the SET_PRESSURE_ADVANCE command - see docs/G-Codes.md for details.

-Kevin

tony978, 2018-11-07:
@KevinOConnor  thank you !

closed 2018-11-07 by tony978