Klipper3d/klipper #804: [FEATURE REQUEST] Box/Chamber temperature

derdigge, 2018-10-21:
Dear All.

I have a printer with enclosure. To monitor the inner temp i installed a termistor inside.
For displaying and tracing i created a "fake" extruder which displays the temp as T1 then in
octoprint. Wouldnt it be nice to have a Box monitor with smaller config?

For examle just a thermistor as analogN may be ad a PWM for Fan or heating. I saw the "my_temp_fan".
maybe it is easier to give this one the ability to report to octoprint?

Thanks
derdigge

Hywelmartin, 2018-10-23:
As klipper uses reprap gcode

https://reprap.org/wiki/G-code#M141:_Set_Chamber_Temperature_.28Fast.29

why not add the whole control to klipper with safety options and alles...with options to control temperature or only cool down(start a fan) and if not set you are able to monitor it..

the quick and dirty sulution would be as you have done and use the next extruder

KevinOConnor, 2018-10-23:
It's fine if someone wants to add support for this.  It isn't something I plan to work on myself though.

-Kevin

KevinOConnor, 2019-02-08:
I'm going to close this for now.  There's already a PR (#855) to track this.

-Kevin

closed 2019-02-08 by KevinOConnor