Klipper3d/klipper #810: Error flashing firmware in Arduino DUE

shadrincev, 2018-10-22:
Hello! I followed all the steps in the installation guide and when I execute make flash FLASH_DEVICE=/dev/ttyACM0 it start to flash but at the end throws src/sam3x8e/Makefile:38: recipe for target 'flash' failed.
The Arduino IDE Repetier Software is perfectly stitched. Tried to sew with erase and reset buttons, and with the soldered resistor at the transistor for additional reset and without it, was stitched on debian and on dietpi, but result one and the same src/sam3x8e/Makefile:38: recipe for target 'flash' failed. The same problem was in the post https://github.com/KevinOConnor/klipper/issues/772 but there is no solution. Please help to solve the problem. 
Thank you!

jakep82, 2018-10-22:
The error can be ignored.  I received the same error, but my Due flashed fine and it works as expected.

KevinOConnor, 2018-10-22:
This appears to be a dup of issue #772.  Lets move the conversation there.

-Kevin

closed 2018-10-22 by KevinOConnor