Klipper3d/klipper #812: Option 'position_endstop' is not valid in section 'stepper_z'

niceddes, 2018-10-23:
Hello together,
 
i have a very strange issue and cannot find the failure. :-(

After connecting from octoprint i always get the following error message:

> Option 'position_endstop' is not valid in section 'stepper_z'

I cannot see a error in the config file or in the log file which i have attached to this post.

Before i got this error message i was trying to setup virtualSD in the config file. Now i changed everything back to the old setup and still get this error message.

Thanks a lot in advance. 

BR

[klippy.log](https://github.com/KevinOConnor/klipper/files/2507994/klippy.log)

jakep82, 2018-10-23:
This was changed a little while back.  When using a probe, you shouldn't define position_endstop for Z.  

https://github.com/KevinOConnor/klipper/commit/c5d4e14298525e0a184f42f812041b32dfea6951

closed 2018-10-23 by niceddes

niceddes, 2018-10-23:
thank you very much!