Klipper3d/klipper #814: st7920 screen not working with Arduino Due

jakep82, 2018-10-24:
I've tested the screen (RepRap Full Graphic) with an MKS Gen 1.4 board flashed with Klipper and it works as expected.  I connected the screen with jumper wires and confirmed the correct pinouts using only the SID, SCLK, and CS pins plus a separate 5V power supply.  The same screen connected to an Arduino Due does not work.  I've tried using multiple pin combinations, but nothing will display an image.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2508891/klippy.log)

FHeilmann, 2018-10-24:
I also tried the same thing on the duet with little success. My hunch is that you need a level shifter for the logic pins since the due and the duet only have 3.3V logic, while arduino megas etc. use 5V logic.

KevinOConnor, 2018-10-24:
I wired my display up to my Due and it works fine for me.  (I used 5 jumper wires and went directly from the display to the due - two wires for 5v/gnd and 3 wires from ar43/ar42/ar47 to cs/sid/sclk.)

@jakep82 - there are a bunch of "extruder0" errors in your log - make sure you aren't trying to test the display while there are errors in the config.

-Kevin

KevinOConnor, 2018-10-24:
[image]

jakep82, 2018-10-24:
I forgot to comment out my extruder fans, but would that make a difference?  I'll try again tonight.

jakep82, 2018-10-24:
@FHeilmann I actually have a RAMPS-FD LCD adapter laying around.  From what I see on the schematic, it should do the level shifting you mention, but Kevin appears to be having success without it.  If I can't get the screen to work wired directly, I'll try the adapter with level shifting and see what happens.

jakep82, 2018-10-25:
Ugh, I hate that it was something so simple!  I spent several days trying to get it working.  Commenting out the fans did indeed fix it.

closed 2018-10-25 by jakep82

ZhenyaKa, 2018-12-07:
KevinOConnor, can you show back of LCD panel? What pins you did use there?

Adel110, 2019-02-17:
R