Klipper3d/klipper #820: Delta Printer: G28 produces move out of range

klHaribo, 2018-10-25:
after last "git pull" (maybe one week ago) my delta Printer is not homing any more.
The error is: "Move out of range: -1.939 -4.578 216.757 [0.000]" (distances are depending on different position_endstop settings)
This error seems to be caused by individual position_endstop settings for each tower.
if I only use one position_endstop (for stepper_a),  the printer homes as expected. But then I am unable to print ;)

[logfiles.zip](https://github.com/KevinOConnor/klipper/files/2516291/logfiles.zip)

KevinOConnor, 2018-10-26:
Alas, you've hit a bug in Klipper.  As a short term workaround, you can add `homing_retract_dist: 10` to the stepper_a section of your config.  I'll put together a fix in the next few days.

-Kevin

KevinOConnor, 2018-10-27:
This should be fixed now (commit 936a4acd).

-Kevin

klHaribo, 2018-10-31:
The last "git pull" solved the problem. Sorry for the delay, I was very buzy this week...

closed 2018-10-31 by klHaribo