Klipper3d/klipper #821: Error: Heater heater_bed not heating at expected rate

aegelsky, 2018-10-25:
Hi

I have 2.5kW heat bed, 1230x550mm, 220V
it won't heat as usual small 12V one, but it does heat.
I can't get it even to 40C now, getting this error in title.
Here's bed config:
[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: zboard:analog14
control: watermark
min_temp: -50
max_temp: 130

yes -50 there is for reason, it can be damn cold in my garage, like -15 -20C for sure :)

any luck getting this bed working with klipper?

Thanks!

aegelsky, 2018-10-25:
also I have a glass upon it, because flat factory heater needs something heavy upon it.
and because everybody uses glass on their printers)
so it takes time to heat glass, especially when it's cold outside (like it is now)

jakep82, 2018-10-25:
Please include a log to help identify any problems.  Klipper has default heater safety settings.  Changes can be made if necessary.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

jakep82, 2018-10-25:
See the verify_heater section in the example-extras.cfg file for more info on the default heater safety values.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg