Klipper3d/klipper #822: Klipper logo

BKLronin, 2018-10-26:
Is there any logo already for the firmware? I could possibly make a simple one.

mamoit, 2018-10-26:
This would be awesome, I want to put a sticker on my Orange pi case, I just need something to put on the sticker :smile:

Hywelmartin, 2018-10-26:
i'm not sure if "Klipper" is some what associated with Clipper(boat type).. but I found this..
[image]

BKLronin, 2018-10-26:
I initially thought about a simple word logo that expresses the heart of klipper with its lines directly. A klipper as minimal graphic would be nice too. Depends on what Kevin might like more. @KevinOConnor

KevinOConnor, 2018-10-27:
FYI, @shoreliner proposed this a few months back:
[image]

theopensourcerer, 2018-10-27:
I prefer the circuit board idea to the ship, but both seem overly complex for a logo.

@kevin why is Klipper called Klipper?

BKLronin, 2018-10-27:
> FYI, @shoreliner proposed this a few months back:
> [image]

Wow, that looks excellent, I would vote for that. I don´t think it will get any better. (I can´t for sure).

frankvdh, 2018-10-27:
I like the idea of the clipper ship and the octopus, linking to Octopi. But
I think a logo needs to be simpler than the above examples, so I did a
search for kraken and clipper, and found the attached, which I think would
be a good starting point...

bruce356, 2018-10-28:
Hi, I think the K looks good but with out the spaghetti and without the second examples white dots surrounding it.
regards - bruce

LazaroFilm, 2018-10-28:
I was also working on a logo on my spare time. Definitely not finished but since the conversation is going... I was also for the Klipper Ship and Kraken tentacles (note, I did this logo before seeing any of this conversation) 

[image]

KevinOConnor, 2018-10-28:
> @KevinOConnor why is Klipper called Klipper?

There is no particular reason for it.  I wanted a name that would be distinct on a web search.  I had a few different names I was thinking about (none of which I now remember) and I just choose one.

-Kevin

BKLronin, 2018-10-28:
I had somethin clear and edgy in mind. Trying to express precision and fine lines, with a light 80ies touch. 
Something like this loose sketch.
[image]

KevinOConnor, 2018-10-31:
Interesting.  I don't have any drawing abilities so I don't have much comment to give on a logo.  My personal preference of the above would be the "K" logo.  Though I agree that the wires outside the "chip" in that logo could be removed.

-Kevin

aeropic, 2018-10-31:
[image]
What about something like that with the K symbolizing the hot end and some filament extruded from it ?
I must admit I'm not an expert in drawing ;-)

aeropic, 2018-10-31:
[image]
Or this one ?

aeropic, 2018-11-01:
Pushing my idea of the K hot end... I propose this one:
[image]

theopensourcerer, 2018-11-01:
FWIW, I like the idea of using the K like a hot-end/extruder. I am wondering though, for a Logo, do you really need to have the "ipper" as well?

I am thinking of, for example, the logo for KDE, e.g.

https://www.kde.org/stuff/clipart/logo/logo-white-blue-source.svg

aeropic, 2018-11-01:
@theopensourcerer , a minimal version for you !
[image]

I like it as well as a logo it's maybe enough ;-)

theopensourcerer, 2018-11-01:
I think we need more feedback and input from others but I for one certainly like the idea.

BKLronin, 2018-11-01:
I´m still a fan of the reduced shoreliner version but aeropics idea is essentially nice as well. 
I minified his version a llittle more. Needs finalizing for sure.
[image]

mgth, 2018-11-01:
I Love the idea of the k hotend, I think the L is too much, maybe something more graphical
[image]

aeropic, 2018-11-01:
@mgth 
You're right the L is too big.
Here is a reduced version
[image]

aeropic, 2018-11-01:
@BKLronin , your hot end is flying too high, you'll get poor adhesion of the first layer ;-)

Avalonnw, 2018-11-01:
May I suggest just k-hotend without L?

aeropic, 2018-11-01:
@Avalonnw  Here it is ... Maybe somewhat too minimalist ?
[image]

Avalonnw, 2018-11-01:
> I read that as Vlipper on the first glance :D

Good logo should be readable in monochrome. 2 colors are good but if you
convert red to blsck it will be unreadable.

blaskkaffe, 2018-11-02:
I have not had time to try my ideas yet but I think it would be nice to somehow show the separation between processing/scheduling and doing the mechanical work which is the bit thing that makes klipper klipper.

One thought I had was half a brain, half cog/hotend/stepper motor. Alternatively if you guys liked the boat idea a thougt was to have a lighthouse/captain/map and a boat for example. I have done some sketches but will try to make something worth showing :p

aeropic, 2018-11-02:
@Avalonnw , you're right
The black & white or grayscale issue is easy to solve.
Either inserting one pixel white around the letter L or painting it into another less saturated color.
Or both !

[image]

aeropic, 2018-11-02:
we could even play with Kevin's initials ;-)
[image]

BKLronin, 2018-11-02:
Next iteration with some of the ideas integrated.
[image]

Avalonnw, 2018-11-02:
Overloaded with small details IMHO

BKLronin, 2018-11-02:
Yes, I had the same impression. Still trying to find out how to express klippers nature with few elements

dragonnn, 2018-11-02:
The last one without the squares would be the best.

BKLronin, 2018-11-02:
I like the squares, expressing the companion idea between raspi and 8 bit but it would like this. 
[image]

aeropic, 2018-11-02:
Not bad... why not enlarging the "Lipper" to something like 
[image]

BKLronin, 2018-11-02:
Like this? 
[image]

aeropic, 2018-11-02:
Yes, left one...
Now, just do one witthout the red squares, enlarge the bottom part of the big L so that it fits the width of the other letters, enlarge the "LIPPER" to fill the square and just try to make the parts of the letters PER overlapping the K in white...

aeropic, 2018-11-02:
Something like that ... but with the overlapping part of the letters PER in white :
[image]

aeropic, 2018-11-02:
With Arial rounded bold police
[image]

BKLronin, 2018-11-02:
Like this
[image]

aeropic, 2018-11-02:
It's becoming very good !
Please just try to find a police that is higher and maybe thinner in width to be more balanced with the tall L ... See above the Arial Rounded bold ;-)

BKLronin, 2018-11-02:
I lik ethe font so far but here some alternatives 
[image]

aeropic, 2018-11-02:
I love the top right one. We can see the deposited filament flowing out of the nozle !

BKLronin, 2018-11-02:
> I liked the top right one too. But, I think my preference would be to remove the part of the "K" that is behind the "per".

Do you mean the "only letter cutout" or the "bigger square cutout"?

BKLronin, 2018-11-02:
[image]

theopensourcerer, 2018-11-02:
Yes, that.

theopensourcerer, 2018-11-02:
The more I look at that the less I like it  - lol... Perhaps then the one you also did where the "per" text became transparent in the bottom of the K - but with this typeface? Maybe that would look better?

BKLronin, 2018-11-02:
[image]

aeropic, 2018-11-02:
Agreed, I don't like neither the cropped nozzle...
Here is another attempt with the overlapping text made white and a filled nozzle so that it becomes more obvious it's a nozzle but the K is still present...
[image]

BKLronin, 2018-11-02:
I had the triangle filled in an internal version but it instantly reminded me of the Kodak logo.

theopensourcerer, 2018-11-02:
Gosh this is harder than it looks to find something that looks "right"... I'm not sure I like the transparent text or the filled in nozzle to be honest. I think something like these is very close but I feel I am not enough of a designer to be able to comment much more.

I guess what we should really do is get some input from Klipper's SABDFL ;-) 

Or several capable people come up with ideas and a vote is held...

theopensourcerer, 2018-11-02:
FWIW - My preferred design so far is this one (although I am not sure it is "right"):

https://user-images.githubusercontent.com/6392076/47913220-940d5900-de9b-11e8-9390-2c3d227e2c98.PNG

aeropic, 2018-11-02:
Which means we are very close to a perfect logo !!!!
A pitty Kodak uses a hot end with a non flat bed ;-)
[image]

Look it works really fine in small like a PC icon
[image]

BKLronin, 2018-11-02:
We don´t have to nail it just now but I think. I pretty much like this one but I´m biased as well.  

[image]

mamoit, 2018-11-02:
Left one!

aeropic, 2018-11-02:
Much nicer in red filament, I agree !!! Don't change anything and let all the others vote ;-)

In red, left one is cool !

theopensourcerer, 2018-11-02:
Left. Looks really good to me. Well done.

BKLronin, 2018-11-02:
Thanks, I prefer the left one as well. :)

aeropic, 2018-11-02:
Just a try with a redFilament
[image]
_full nozzle

BKLronin, 2018-11-02:
A bit to heavy for me. I like the slightly "hidden" nozzle.

mamoit, 2018-11-02:
Can you try the previous version (non-full) but connect the top of the 'L' to the red outer part of the 'K', as if the filament was going through the extruder hole.

[149 more replies]

closed 2019-05-25 by KevinOConnor