Klipper3d/klipper #823: Ability to change probe offset with SAVE_CONFIG

mamoit, 2018-10-26:
Every time that I move the probe a little along the Z axis, I have to manually change the `z_offset` on the probe.
Is it a good idea to use the new `SAVE_CONFIG` feature to save the Z component of `SET_GCODE_OFFSET` to the config file?
Or would it be better to have some other calibration command for the probe specifically?

KevinOConnor, 2018-10-27:
I'm not sure what's being reported and/or requested here.  If you've manually altered the physical location of the probe then you'd need to recalibrate the probe.  I don't see any way to avoid that.

-Kevin

filippor, 2018-10-27:
I propose a calibrate probe. Probe with automatic probe and manual move to zero

mamoit, 2018-10-27:
Let me rephrase it then.
SAVE_CONFIG is awesome since it allows us to save the configuration files without needing to edit the cfg file on a terminal. This opens the door for scripting our way into automating the calibration of any part of the printer (from xyz max to probe offset).
From my understanding it currently only saves the bed leveling variables, but it could potentially be used to tweak anything on the menu system without digging into the configuration file.
For this, instead of SAVE_CONFIG we could have a SET_CONFIG of some sort.
Sorry if I can't express myself clearly.

KevinOConnor, 2018-10-28:
I think a PROBE_CALIBRATE command would be useful.  I've considered adding it myself, but I'm interested in seeing how we can better integrate the menu during manual probes before adding more manual probing code.

I personally don't think a generic SET_CONFIG command is needed.  I think config editing tools (like the ssh editor plugins or the web based one in OctoKlipper) provide a better all around solution.  I suspect editing the config via a g-code terminal session would not be particularly user friendly.

-Kevin

LazaroFilm, 2018-10-29:
I would love a probe calibrate where you could add or remove to the value incrementally maybe something like PROBE_CALIBRATE -0.05 would decrease the probe offset by 0.05

Avalonnw, 2018-10-29:
I vote for Probe_calibrate!
1. Set zero at probe_xy position manually using paper method.
2. Send NEXT. Nozzle lifts probe_clearance_z
3. NEXT. Attached probe (or automatically deployed one) moves down and gets triggered probe_avg times. Probe offset gets calculated and stored into config.

andremariz, 2018-11-01:
Im sorry if this os the wrong place to ask but since you are talking about probin;
I want to try klipper but i have my printer fitted with ABL. 
How can i use it in Klipper?