Klipper3d/klipper #824: Firmware retraction

nythil, 2018-10-26:
Have you considered adding firmware retraction gcodes (G10/G11)? I find it easier to change retraction settings in firmware than in the slicer. In fact, in Slic3r retraction settings are tied to printer settings, while it is possible to add custom gcode to tune FW retraction for each filament separately.

FHeilmann, 2018-10-26:
Until it gets added you can sort of work around it by defining two gcode macros that do the retract and in retract

nythil, 2018-10-26:
I'll try that. Thanks for the idea! Also, I can probably add firmware retraction myself - I mostly wanted to know if the idea was acceptable for klipper

speendo, 2018-10-30:
@nythil Did you write the macros?

If yes, would it be possible that you provide them here? I just installed klipper and these macros would be of great use to me!

Thank you!

Lenbok, 2018-10-31:
I find that I vary the retract/unretract distance depending on the filament I am using (e.g. stiff PLA short retract; ABS medium retract, Nylon large retract). With my marlin and smoothie printers I use M207 to accomplish this (usually this is in the slicer filament-specific gcode section, but sometimes I real-time adjust while tuning for a new filament).

Is this the sort of thing that could be accomplished with the macro method? (i.e. is it possible to write a M207 macro that takes parameters, to be retrieved by the G10/G11 macros?)

speendo, 2018-10-31:
I'm far from understanding all features that Klipper - or even the config files - offer.

But my first thought is, that besides gcode macros we would need something like "gcode variables" - like

    [gcode_var M207]
    name: RETRACTION_LENGTH
    default_val: 2.5

With that my code above could look something like

    [gcode_macro G10]
    gcode:
        G91
        G1 E-{RETRACTION_LENGTH} F4000
        G90

nythil, 2018-11-03:
@speendo is right - gcode macros seem to be missing variables that could be modified on-the-fly. I'm going to implement a proposal for a firmware retraction module.

LazaroFilm, 2018-11-05:
This is making me thing of something slightly different. I feel like a lot of parameters linked to the material are currently determined by the slicer's gcode. This forces us to reslice an object if we are changing material. I feel like those settings could be loaded in a separate "material" or "filament" file that you would select when starting a print. So you would select 1. your sliced model, 2. your material before a print. 
The material parameters could be:
Temperature, speed, retraction. 
This might be too ambitious for Klipper only, but I feel like the G-code system is very archaic and could definitely be enhanced by this kind of system.

manu7irl, 2018-11-05:
I think this is the next big thing to embbed in klipper the materail
properties like what you said but also Esteps and even more maybe let the
gcode parser to slpit the code like only the exterior walls would be like
multi-material or multi-color based on the calculated hotend position and
for infill it will be only one material or the purge for the material
transitions.

trevjonez, 2018-11-07:
you can add custom python logic and have the gcode macro run a function in python code. I am using it on my prusa mk3. there is an example in the prusaowners/klipper fork. https://github.com/PrusaOwners/klipper/blob/master/klippy/extras/custom_gcodes.py

also just wanted to add a note here. I am seeing some nylon filament expand during move operations just after a retraction. would be nice to dynamically adjust the deretraction length based on travel time so I could tune the extruder behavior to the material properties. I am able to emulate the behavior thus far with tiny negative extra restart lengths, but the longer it sets in the nozzle not extruding the more it oozes.

Lenbok, 2019-05-07:
I've opened #1617 implementing firmware retraction with runtime parameter adjustment.

IgorKokovkin, 2020-01-22:
tried to set 
[gcode_var M207]
name: RETRACTION_LENGTH
default_val: 2.5

got error
ParsingError: File contains parsing errors: /home/pi/printer.cfg
	[line 180]: '    name: RETRACTION_LENGTH\n'
	[line 181]: '    default_val: 2.5\n'

see details in klipper.log

FHeilmann, 2020-01-22:
This format was a suggestion on how to implement it. The suggestion was not accepted, which is why you are getting those errors. The code is not there.

IgorKokovkin, 2020-01-22:
Got it, thanks