Klipper3d/klipper #827: LCD babystepping support

lf-, 2018-10-28:
Babystepping in its current implementation is really hard to get perfect. It's great that it's supported at all to baby step, but the current implementation is really not usable for printers which have both lots of warp and Z probes with unpredictable offset. What I used to did on Marlin was to double click the encoder to get the baby step entry, then tune it in at the beginning of prints with the encoder wheel while looking at the print of some first layer infill or a skirt. Having to do the offset from OctoPrint makes this workflow nearly impossible.

If babystepping/axis offset were adjustable from the LCD, this would be much easier.

bruce356, 2018-10-28:
@lf- 
Comment by Janar - Btw you can now have babystepping in realtime with exact values.
There's new gcode.homing_zpos parameters available.
See the new Tune menu item Offset Z, visible only in printing time.
```
[menu __tune __offsetz]
type: input
name: "Offset Z:{0:05.3f} "
parameter: gcode.homing_zpos
input_min: -5
input_max: 5
input_step: 0.005
realtime: true
gcode: SET_GCODE_OFFSET Z={0:.3f}
```

lf-, 2018-10-28:
Oh, neat. So it's essentially handled already. I'd still like the double click to get it, but this is excellent news.

bruce356, 2018-10-28:
You can not necessarily have everything at once

Boffgithub, 2018-11-02:
> .... Having to do the offset from OctoPrint makes this workflow nearly impossible.
> 
> If babystepping/axis offset were adjustable from the LCD, this would be much easier.

If you are babystepping in OctoPrint by typing the extended gcode command each time, then there is an easier way:-  Create a button for each command using the TerminalCommands plugin for OctoPrint.

E.g. I use:
'Babystep /\'  = SET_GCODE_OFFSET Z_ADJUST=0.2
and
'Babystep \/'  = SET_GCODE_OFFSET Z_ADJUST=-0.2
Which work fine for live adjustment during the skirt or 1st layer

Boffgithub, 2018-11-02:
actually 0.02 but you get the idea!

KevinOConnor, 2018-11-17:
I'm not sure what the state of this issue is.  If it can be implemented in the menu, should I close this issue then?

-Kevin

lf-, 2018-11-17:
Yes, it can probably be closed. I couldn't quite get it working with the option in the root menu rather than tune, but I think that was just user error.

closed 2018-11-23 by KevinOConnor