Klipper3d/klipper #828: Delta Calibrate: Documentation incomplete

JustAnother1, 2018-10-28:
There is no information available in the documentation how the Basic delta calibration can be executed in manual (paper test) mode.

After issuing "G28" followed by "DELTA_CALIBRATE METHOD=manual" the printer moves to the first position and just stops. I assume that at this point of time the user somehow needs to tell the printer if the nozzle is too high or not. The Terminal does not show any instructions of how to do that. There is also no information in the document https://github.com/KevinOConnor/klipper/blob/master/docs/Delta_Calibrate.md of how to do that.

The next step is probably to issue a "NEXT" command to move to the next position. This is also not documented anywhere. Or did I miss the part of the documentation that explains this?

Thank you for your help!

Avalonnw, 2018-10-29:
When it stops, jog it down to catch the paper and send NEXT. Repeat.

JustAnother1, 2018-10-29:
What does "jog" mean? Does it mean sending "G1 Z0.5" and similar commands? Or is there an easier way to jog?

Avalonnw, 2018-10-29:
Yes. Just use relative motion and go slow. Or, on motion tab in octoprint use arrows (up-down).

JustAnother1, 2018-10-30:
Thank you for the explanation.

So the next step would be to add the information to use G-Codes on the terminal to move the nozzle towards the bed and to then issue a "NEXT" command to the "Delta_Calibrate.md" documentation file, right?

JustAnother1, 2018-10-30:
I played some more with it:

- Octoprint seems to not do movements below  Z0. Even though it switches between relative and absolute coordinates once you hit Z0 the button to move down some more just does nothing. Seems to be a safety measure in the buttons. If the command is issues in the octoprint terminal window then movements to Z-1 are OK. It would probably be good to document this behavior. 

- "SAVE_CONFIG" is a bit strange. I specify the configuration file once I start Klipper. the "SAVE_CONFIG" did not change the configuration file I use. But it created a a configuration file /home/pi/printer.cfg. That file is an copy of the configuration I use, but it has "position_endstop" and "arm_length" commented out and added a commented out block at the end of the file. There seems to be some parsing magic used to extract the new values from the commented out section. Wouldn't it be better to actually update the settings instead of commenting them out?

Avalonnw, 2018-10-31:
1. I doubt it's anything to do with Octoprint. What do you have in minimum_z_position parameter?
2. Annoying, yes. There are previous configs stored actually with time and date stamp in the root folder also. They get generated every time you run save_config.

JustAnother1, 2018-10-31:
@Avalonnw 
1) I have (as in the documentation) added the "minimum_z_position = -5". I assume it is octoprint, as the buttons (as in your comment) do not work. Terminal input in octoprint works. So for some reason the octoprint button does not generate the commands. Maybe someone else can verify this?

JustAnother1, 2018-10-31:
@Avalonnw Thanks for verifying. Then it probably is just me.

KevinOConnor, 2018-10-31:
I do plan to extend the documentation for manual probing (not just for delta_calibrate, but also for bed_tilt_calibrate / bed_mesh_calibrate).  There's been some work recently to get the lcd menu (if present) into a mode that will help the user perform manual calibration and I'd like see how that work turns out.

The reasons behind the save_config behaviour are described in pr #666.

-Kevin

JustAnother1, 2018-11-05:
Regarding the save config issue:
As Klipper is able to read the values from the configuration file it must be possible to do a replace with the new value.
Basically finding the line that actually configures the value (so detecting sections, ignoring commented lines) as it does when it reads the configuration file.
When reading the configuration the value is then extracted from the configuration file line. In this step the value needs to be replaced in the line.
Do this for all settings updated from pidTune/BedLeveling/,.. and then save the modified file.

It is probably not easy to do, but doable. My python foo is sadly not yet strong enough to implement this myself. But the changes would probably go in here: https://github.com/KevinOConnor/klipper/blob/master/klippy/configfile.py

It would be great if someone could implement it.

KevinOConnor, 2019-02-12:
FYI, there is now further documentation on manual calibration (commit 57e6acbc).

-Kevin

closed 2019-02-25 by KevinOConnor