Klipper3d/klipper #831: [INFO] Direct stepper driver control from RPi SPI

dpellegr, 2018-10-29:
Sorry if this has been discussed before, I made extensive searches but I couldn't clarify my doubts.

I was wondering if it would be possible to use the SPI controller of the RPi3 to directly talk to SPI-enabled stepper drivers (eg TMC2130) without the need of an intermediate AVR.

Is this available/implementable/impractical in Klipper? Great Thanks!

jakep82, 2018-10-29:
> Is this available/implementable/impractical in `Klipper?

No/No/Yes

dizzey1, 2018-10-29:
I think that driving steppers directly from the pi is not a great idea since the pie is not running a real time os. there are ways around it but i dont think they are practical, you could do the driver in kernel space and bind a core or take realtime priority but the rest would suffer. You could run a realtime linux but then you need to rewrite klipper from scartch. But then the whole point of klipper is to let the cheap mcu handle hard realtime demands but still let you do all the heavy liffting on the fast pie

KevinOConnor, 2018-10-29:
If you are looking to fully control the motion of a stepper motor directly from a Raspberry Pi, then the current code is not particularly well suited to do that.  It's theoretically possible but would require a great deal of work.  I suspect it would not be worth the effort.

If you are looking to just configure a stepper motor from the Raspberry Pi then that may be possible, but it would require some unix admin knowledge and may require some python code changes.  You would also need to ensure the devices had the same voltage level - the rpi is 3.3v while common AVR devices use 5v.  Assuming voltage levels match the printer could (theoretically) be configured with the following steps: compile and install the "linux" micro-controller target, update your printer config to use multiple micro-controllers, and configure the tmc drivers to use the spi bus on the "linux mcu".  For reference, the replicape boards effectively do this in order to control all the peripherals on that device.

-Kevin

LazaroFilm, 2018-11-05:
You could connect your stepper motor to a smaller Arduino like the Arduino Beetle if you only need to control a single motor. 
https://www.aliexpress.com/item/Beetle-Virtual-Keyboard-BadUSB-Pro-Micro-ATMEGA32U4-Module-Mini-Development-Expansion-Board-For-Arduino-Leonardo-R3/32847715254.html