Klipper3d/klipper #832: Velleman K8200 config file. SOLVED

PerLangkjaer, 2018-10-29:
I have used some time and solved the config file for the K8200/3Drag printer.

[Velleman8200printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/2526783/Velleman8200printer.cfg.txt)

dizzey1, 2018-10-31:
the promotion video was released "Published on Apr 15, 2013" https://www.youtube.com/watch?v=MNEZLUXctf8&t=24s  but you can still order them

PerLangkjaer, 2018-11-04:
@KevinOConnor 
You shoukd put this code into the examlple list of the build.

KevinOConnor, 2018-11-13:
Thanks.  I pulled this in as an example config (commit 22f7be8b).  I made some changes to the config as part of committing (fixed an error in thermistor definition, standardized the comments to be more similar to the other examples, removed the pressure advance settings which are very filament specific, and removed the slicer specific settings).

-Kevin

closed 2018-11-13 by KevinOConnor