Klipper3d/klipper #833: [Discussion] Relaxing the restrictions on SX1509 pins

FHeilmann, 2018-10-29:
Hi @KevinOConnor,

thanks again for merging the SX1509 support. Me and a few fellow duet users have been testing the extra and the following 'issue' arose: The way the extra is configured right now prohibits the use of an SX1509 pin as a heater, which is how you and I intended it. However, it also prohibits the use of SX1509 pins as `temperature_fan` or `heater_fan`. I think those two applications are perfectly fine for a SX1509 pin, since, in the worst case, the fans would not be cooling a properly shut-down heater, which isn't ideal, but not threatening.

It seems that heaters can be uniquely identified by their non-zero max_duration value, which means we could potentially remove the check in the SX1509 extra here:
https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/sx1509.py#L129
https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/sx1509.py#L193

Technically this would mean that sx pins do not respect a shutdown value, however, as I described above, I think this is acceptable so long as the pin is not a heater.

another option woudl be to have some sort of "I know what I'm doing" configuration variable which disables this check only if the user knowingly sets it.

What do you think?

Florian

KevinOConnor, 2018-10-29:
I'm fine with ignoring the shutdown value on sx1509.

-Kevin

FHeilmann, 2018-10-30:
Great, I'll submit a PR

Florian

closed 2018-10-30 by FHeilmann