Klipper3d/klipper #838: Firmware crash at random moment

TriBallscom, 2018-10-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2531587/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/2531611/printer.txt)

When I startup the printer I do 
STEPPER_BUZZ STEPPER=stepper_a
G91
G28 Z0
G90
DELTA_CALIBRATE

Before I start a print.
Sometimes DELTA_CALIBRATE makes klipper crash but other moments it crashes after heating the bed and tool.

My printer is a FLSUN Delta (with a arduino mega2560/Ramps 1.4 clone board) and OctoPrint 1.38 (OctoPi) is running on a RPI2B.

I have attached my (renamed) printer.cfg and klippy.log

FHeilmann, 2018-10-31:
`MCU 'mcu' shutdown: Timer too close`
that's the error you're getting. looking at your printer, 200mm/s `max_z_velocity` sounds like a LOT. I'm just guessing but try setting this value to what is suggested in the example-delta.cfg (`max_z_velocity: 150`) and see if that fixes it.

KevinOConnor, 2018-10-31:
This problem is due to the use of a slower host cpu (a raspberry pi1?) and a single long move (G28 ; G1 Z15.0 F9000).  That move requires scheduling ~80K steps and sometimes the host can't calculate all those steps before exceeding its scheduling deadline.  Changing the speed of the move wont help.  Currently the only workarounds are to use a faster host cpu or to break up that really long move into two moves (eg, G28 ; G1 Z100 ; G1 Z15).

-Kevin

TriBallscom, 2018-10-31:
Hi Kevin,

Thanks for your response.
I understand that the RPI 2 might be a challenge but I have a rpi 3 laying around, would that do or would you suggest a faster host?

Ron

TriBallscom, 2018-10-31:
Setup the RPI3, seems much better..

closed 2018-10-31 by TriBallscom

KevinOConnor, 2018-10-31:
Either a rpi2 or rpi3 should be fine.

-Kevin

TriBallscom, 2018-11-01:
I used a RPI2b that did not work that great, now the RPI3 is working much better.

Thanks for your help.

Ron