Klipper3d/klipper #839: Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2581

Devilcow, 2018-10-31:
Printing with a Creality CR-10S with the v2.1 mainboard
OctoPrint version : 1.3.9
OctoPi version    : 0.15.1
RaspberryPi 3+

I was able to flash the firmware using the /dev/ttyUSB0, and this is what is pointed to in my /dev/serial/by-id/usb*

I also have the OctoKlipper plugin installed on my octroprint server.
I've attached my logs, but cannot find anything that looks too  useful to me... Is this an issue of configurations?

The following is my printer.cfg
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^ar18
position_endstop: 0.5
position_max: 300

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: pid
pid_Kp: 690.34
pid_Ki: 111.47
pid_Kd: 1068.83
min_temp: 0
max_temp: 130

[fan]
pin: ar9

[mcu]
serial: /dev/ttyUSB0 
baud: 250000
pin_map: arduino
restart_method: arduino

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 3000
max_z_velocity: 25
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17
encoder_pins: ^ar33, ^ar31
click_pin: ^!ar35

[serial.zip](https://github.com/KevinOConnor/klipper/files/2532101/serial.zip)
[octoprint.zip](https://github.com/KevinOConnor/klipper/files/2532102/octoprint.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2532103/klippy.zip)

Devilcow, 2018-10-31:
Sorry, while cleaning up my config file to share here I accidentally deleted one of the required fields, the signal pin or some such on the extruder, I have since fixed that and confirmed again that this issue is still persisting.

FHeilmann, 2018-10-31:
You don;t connect to the device serial port directly with octoprint, that's what the klippy process has to do. You have to connect to klippy's virtual serial port (default location is `/tmp/printer`)

Check this section of the installation instructions:https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

Devilcow, 2018-10-31:
Sadly I've tried that. I've iterated through using /tmp/printer as the serial port in Octoprint, and even using it in the printer.cfg file. 
I've also tried setting the baud rate to 115200, just to make sure that was not an issue... 
Any other ideas?
Thanks!

Devilcow, 2018-10-31:
Slight change to the situation, I have re-flashed the bios on the MCU and now, I'm not getting the serial port error anymore, but now I am getting a time out... I have upped my communications timeouts significantly, is there somewhere else, perhaps in the config file I can set the timeout?
[serial.zip](https://github.com/KevinOConnor/klipper/files/2533684/serial.zip)

Devilcow, 2018-10-31:
Please disregard, I found my problem, I had the wait for Start option checked in the firmware settings on octoprint. 
Thanks for the assistance!

closed 2018-10-31 by Devilcow