Klipper3d/klipper #840: Klipper firmware upload using serial port fails

sgoum, 2018-10-31:
Hi
I'm trying to install Klipper in a MKS GEN L board.
The board connects to RPi using direct serial access and not USB.
I enter:
make flash FLASH_DEVICE=/dev/ttyS0
and receive:
timeout message.
Any idea what am I doing wrong?

Thanks

theopensourcerer, 2018-10-31:
The looks like the wrong device name completely...

Please see: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

sgoum, 2018-10-31:
pi@octopi:~/klipper $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 5 Oct 30 17:17 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 Oct 30 17:17 /dev/serial1 -> ttyAMA0
pi@octopi:~/klipper $

Gomer05, 2018-10-31:
I have a MKS Gen L board and use the USB port.  When you say "serial" are you doing something special?
Follow the steps in the docs noted above to find your PORT,  you can also do a 'dmesg' or check /var/log/messages for info as well.

On my RPi2 I get the following in /var/log/messages when plugging in or removing the USB connection to the MKS Gen L board.

Oct 31 07:30:24 octopi kernel: [1490108.924443] usb 1-1.5: new full-speed USB device number 37 using dwc_otg
Oct 31 07:30:24 octopi kernel: [1490109.057732] usb 1-1.5: New USB device found, idVendor=1a86, idProduct=7523
Oct 31 07:30:24 octopi kernel: [1490109.057752] usb 1-1.5: New USB device strings: Mfr=0, Product=2, SerialNumber=0
Oct 31 07:30:24 octopi kernel: [1490109.057769] usb 1-1.5: Product: USB2.0-Serial
Oct 31 07:30:24 octopi kernel: [1490109.059357] ch341 1-1.5:1.0: ch341-uart converter detected
Oct 31 07:30:24 octopi kernel: [1490109.061384] usb 1-1.5: ch341-uart converter now attached to ttyUSB0
Oct 31 07:30:24 octopi mtp-probe: checking bus 1, device 37: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
Oct 31 07:30:24 octopi mtp-probe: bus: 1, device: 37 was not an MTP device
Oct 31 09:39:20 octopi kernel: [1497845.457154] usb 1-1.5: USB disconnect, device number 37
Oct 31 09:39:20 octopi kernel: [1497845.458334] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
Oct 31 09:39:20 octopi kernel: [1497845.458456] ch341 1-1.5:1.0: device disconnected

KevinOConnor, 2018-11-07:
When using hardware serial via the rpi gpio pins the required port is usually /dev/ttyAMA0 .  Beyond that tip, I don't have any additional advice.

-Kevin

closed 2018-11-16 by KevinOConnor