Klipper3d/klipper #841: Klipper with RAMPS-FD

protworks, 2018-10-31:
Hi there, guys!

I'm thinking of using RAMPS-FD + Due+ klipper in my printer.
Does anyone have config file of ramps fd for klipper ? 

Thanks for any feedback.
Cheers.
PW.

jakep82, 2018-10-31:
Commercially available RAMPS-FD boards are not safe to use.  See the warning at the link.

https://reprap.org/wiki/RAMPS-FD

protworks, 2018-10-31:
Yes I know it. But I will do the neccessary modifications to make it safe.

jakep82, 2018-10-31:
The design is open source, so you should be able to determine the pins.  Schematics are available on Github.

https://github.com/bobc/bobc_hardware/blob/master/RAMPS-FD/RAMPS-FD-Schematic.pdf

bruce356, 2018-11-01:
hi, or simply copy them from marlin pins files also Ramps-FD v2.1a is available use translator on Chinese web site, RADDS is also a good option, but I hate where they located the dip switches, cant access them with the Due in place.
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.3eab2e8dJxxBxR&id=544866465374&_u=p3qea196e529
regards - bruce

boelle, 2018-11-08:
> Commercially available RAMPS-FD boards are not safe to use. See the warning at the link.
> 
> https://reprap.org/wiki/RAMPS-FD

not true, the first versions of the boards yes, but not the latest ones

boelle, 2018-11-08:
> The design is open source, so you should be able to determine the pins. Schematics are available on Github.
> 
> https://github.com/bobc/bobc_hardware/blob/master/RAMPS-FD/RAMPS-FD-Schematic.pdf

wrong link, this one has newest version (and prob the last) https://github.com/bobc/RAMPS-FD/tree/dev/RAMPS-FD

boelle, 2018-11-08:
the short story is that Geeetech jumped at the first few versions and spammed the market full of them, even thou they knew it was not safe.

the result is that Ramps-FD has got a bad name to it.

and @bobc does not have a very good reason to keep working on it anymore

he did start adding support for TMC drivers but it was not completed.

now everyones pipe dream: support for TMC2130 and TMC2808. Not going to happen as no one has taken up the task

next part is to test and produce them at a GOOD quality and low price, low price is easy enough if you just use junk parts on the board

jakep82, 2018-11-08:
Note that I said commercially available.  Everything I see on Aliexpress is a v1 board, and even bobc himself has told people not to buy them.  I think it's dangerous to ask average users to try and distinguish between versions, especially when so many of the unsafe version are still floating around.  I'm not knocking the RAMPS-FD, but I think it's unwise to tell people they are a viable option unless that person wants to make their own PCBs from the latest version.

jakep82, 2018-11-08:
For what it's worth, I have three v2.1 RAMPS-FD PCBs sitting in a box, and some day I plan to populate them and use them.