Klipper3d/klipper #842: Temperature not remaining stable for a few seconds

dushyantahuja, 2018-11-01:
Please see attached log and screenshot from Octoprint. The temperature goes down to almost 5 degrees below setpoint while printing. It stays stable most of the time - but there are dips.
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2537787/klippy.log.zip)
[image]

Not sure why this is happening - had never seen this when I was running smoothieware. The board is a cohesion 3D ReMix (http://cohesion3d.com/cohesion3d-remix/) with an e3D Titan.

manu7irl, 2018-11-01:
> Please see attached log and screenshot from Octoprint. The temperature goes down to almost 5 degrees below setpoint while printing. It stays stable most of the time - but there are dips. [...]

did you try PID calibration?

`Calibrate PID settings
Klipper supports PID control for the extruder and bed heaters. In order to use this control mechanism it is necessary to calibrate the PID settings on each printer. (PID settings found in other firmwares or in the example configuration files often work poorly.)

To calibrate the extruder, navigate to the OctoPrint terminal tab and run the PID_CALIBRATE command. For example: PID_CALIBRATE HEATER=extruder TARGET=170

At the completion of the tuning test run SAVE_CONFIG to update the printer.cfg file the new PID settings.

If the printer has a heated bed and it supports being driven by PWM (Pulse Width Modulation) then it is recommended to use PID control for the bed. (When the bed heater is controlled using the PID algorithm it may turn on and off ten times a second, which may not be suitable for heaters using a mechanical switch.) A typical bed PID calibration command is: PID_CALIBRATE HEATER=heater_bed TARGET=60`

dushyantahuja, 2018-11-01:
Yes. As you can see from the screenshot, the temperature is stable for most
of the print, except for sudden dips.

BlackStump, 2018-11-02:
have you tried a new extruder heater? ....I have experienced this before that a replacement heater fixed .
I initially thought it was a poor connection/ broken wire but because it was intermittent it was difficult to pin down ....so went the sledge hammer approach of just replacing heater. I doubt it is a firmware issue.

james-golden, 2018-11-03:
One problem that I had with mine was a loose thermistor in the heater block.  I would get problems like not heating quick enough and some times some good dips in the temps and some times the heat would jump up several degrees also.  I now put the thermistor in as far as I can and then add a dab of high temp RTV.  I works pretty well and I have had no problems since doing this.  Don't fill the whole thermistor hole with the rtv, if you do, you might need a new heater block.  Been there and done that.

KevinOConnor, 2018-11-07:
@dushyantahuja - it does look like Klipper could have done better.  Prior to the dips Klipper turns off the heater for a few seconds, and then it needs to turn the heater full on for many seconds to recover.  This looks like it is due to the extremely noisy adc sensor on the lpc176x chips.  (As a guess, a severe noise event results in a distortion of the temperature slope measurement, which results in an undesirable amplification of the Kd parameter.)

Increasing the heaters's smooth_time may help.  A code change to use a "median filter" in Klipper may ultimately be necessary, however.

-Kevin

klHaribo, 2018-11-13:
I had the same problem.
changing heater and thermistor solved it for me.

BlackStump, 2018-12-18:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L959

BlackStump, 2018-12-18:
the run of the mill chinese thermistor if that is what you are looking for
sensor_type : NTC 100K beta 3950
that info is in the example.cfg
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L152

gabi0110, 2018-12-19:
@BlackStump thank you for answer but i don't think that's the same thermistor. In fact they don't look the same and I don't know if they have the same temperature ramp.

KevinOConnor, 2019-01-04:
FYI, I created a development branch that implements a median filter on temperature reports.  I think this should fix the issue with noise on the smoothieboards.  To test the branch:

cd ~/klipper ; git fetch ; git checkout origin/work-median-filter-20190104 ; sudo service klipper restart

and update the smooth_time parameter in printer.cfg to a value of 2.7 (see the updated config/generic-smoothieboard.cfg for further details).

Let me know of your results.
-Kevin

KevinOConnor, 2019-01-14:
The latest mainline code now implements a median filter in the lpc176x micro-controller code (commit d3ec1038).  It is currently believed that this additional feature will resolve the problems raised in this issue.

It is recommended that anyone using an lpc176x board upgrade to the latest code, recompile the micro-controller code, and reflash the board.

-Kevin

closed 2019-01-21 by KevinOConnor

dushyantahuja, 2019-01-25:
Thanks a lot - the latest update solves this issue completely. Apologies, I was not able to test earlier. My printer was down as I had managed to break the Z-axis fittings (I'm using a belted Z-axis).

[image]

Is this still recommended:

[static_digital_output ground_unused_analog_pins]
pins: !P0.2, !P0.3, !P0.25, !P0.26, !P1.30, !P1.31

?

KevinOConnor, 2019-01-25:
>Is this still recommended:
[static_digital_output ground_unused_analog_pins]
pins: !P0.2, !P0.3, !P0.25, !P0.26, !P1.30, !P1.31

No - other testers reported that it did not have a substantial impact.  Updating to the latest code (with the median filter) is the only thing needed.

-Kevin