Klipper3d/klipper #843: [MKS S-Base] Optical endstop need a pulldown resistor

makz27, 2018-11-01:
### Bug report

**Hardware :** MKS SBase (LPC1768) + Optical endstops

**Description :** Optical endstop stay triggered after first trigger
**Expected behaviour :** Endstop is *open*, Home first time, Endstop is *TRIGGERED*, Retract, Endstop goes back to *open*
**Actual behaviour :** Endstop is *open*, Home first time, Endstop is *TRIGGERED*, Retract, Endstop stay *TRIGGERED*

**Terminal log :** 

```
Recv: // Klipper state: Ready
Send: M105
Recv: ok T0:20.7 /0.0 B:20.5 /0.0
Send: G91
Recv: ok
Send: G28 X0 Y0
Send: G90
Recv: !! Endstop x still triggered after retract
Recv: ok
Recv: ok
Send: M112
```

**Marlin issue w/ bugfix for reference :
[klippy.log](https://github.com/KevinOConnor/klipper/files/2538369/klippy.log)
** https://github.com/MarlinFirmware/Marlin/issues/9547

LazaroFilm, 2018-11-05:
I had that issue when I switched from Ramps to MKS Gen-L on my printer. My probe would not work anymore with the dow resistors, so instead, I used a 24 to 5v pulldown and it's working flawlessly now.

KevinOConnor, 2018-11-07:
It's possible to add support for pull down resistors to Klipper.  It does require both host and micro-controller code changes.

-Kevin

mattthebaker, 2019-02-01:
I've dealt with some endstop and probe issues on the SBase before.

If I recall correctly, the SBase has a 1k resistor to 3.3v on all of the endstop inputs.  I think it was an attempt to make it safer for using with 5V inputs to the 3.3V LPC MCU.

If the pull down in the endstop is not strong enough (a mech switch is effectively a short), it does not pull the input to a low enough voltage.  A 3.3v cmos input low voltage is around 0.9v vs 1.5v on a 5v mcu.  On an SBase, the endstop further has to overpower that 1k to 3.3v.

The SBase additionally has a resistor from the endstop input to the MCU.  When using the mcu built in pullup or pull down, this forms a voltage divider and diminishes the effect of the pull.

Turning on the mcu pull down might shift your low value by 100s of mv which may or may not be enough.

When I used optical endstops, I had to change a resistor value on the endstop pcb to get it to pull low enough. When I used an inductive sensor, I had to add a pull down resistor to the input like you describe.

In my opinion, SBase has a hardware design defect in the endstop circuits.

KevinOConnor, 2019-02-27:
I've added support for enabling micro-controller pull down resistors (for those micro-controllers that support it) - commit dddfb681.  To use this support, use a '~' instead of a '^' in the printer config file.  This  support will remain undocumented for now - I'll look to update the documentation in a few weeks.

-Kevin

KevinOConnor, 2019-03-18:
As of commit 53290664, pull-down resistors is now documented.

-Kevin

closed 2019-03-18 by KevinOConnor